OpenDroneMap/ODM #278: Building ODM in Docker: Bundler issue

avstjohn, 2016-03-21:
I'm on Ubuntu 14.04 LTS, and have an issue with Bundler.

The Docker build gets just past configuring and building Ceres, and then:

> ...
>   > ceres
>     - configuring ceres
>     - building ceres
>   < done - Sun Mar 20 21:44:16 UTC 2016
> 
>   > bundler
> ./install.sh: line 407: cd: /code/src/bundler: No such file or directory
> The command '/bin/sh -c ./install.sh &&     chown -R odm:odm /code' returned a non-zero code: 1

What I had to do to make it work was add the following to line 193 of install.sh:

`git clone  https://github.com/snavely/bundler_sfm.git $BUNDLER_PATH`

The script defines $BUNDLER_PATH and related variables, but never fetches the archive, which starts at line 167.

The script finished, and it is behaving thus far on some sample datasets.

dakotabenjamin, 2016-03-21:
Bundler is now depreciated, and run.py defaults to using OpenSfM over bundler. What looks like is happening is the git submodule that is used to download bundler wasn't working. see https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/.gitmodules and https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/install.sh#L154

dakotabenjamin, 2016-05-16:
@avstjohn Docker has been updated, let me know if there are more issues.

closed 2016-05-16 by dakotabenjamin