OpenDroneMap/ODM #281: OpenSfM claims missing GPS info

titanous, 2016-03-24:
I have a few hundred images that are all tagged properly, but OpenSfM refuses to use the tags for finding neighbors with:

```text
WARNING:__main__:Not all images have GPS info. Disabling matching_gps_neighbors.
```

I manually verified that all of the EXIF JSON files that it is using have GPS coordinates in them.

dakotabenjamin, 2016-03-24:
what do the exif tags look like? If they are not in the right projection then I think they will fail

titanous, 2016-03-24:
From `exiftool`:

```text
GPS Altitude                    : 135 m Above Sea Level
GPS Latitude                    : 39 deg 44' 52.92" N
GPS Longitude                   : 75 deg 42' 20.93" W
GPS Position                    : 39 deg 44' 52.92" N, 75 deg 42' 20.93" W
```

dakotabenjamin, 2016-03-24:
That looks fine. You should probably submit an issue over at [OpenSfM](https://github.com/mapillary/OpenSfM), because if this is a problem, it would be with their code. 

First I guess you should try running OpenSfM directly on your images, though, just to be sure.

titanous, 2016-03-24:
It looks like this was fixed upstream: ﻿https://github.com/mapillary/OpenSfM/commit/21735be17768452b94359795aeee9dd524c5b6ab

Can someone update OpenSfM or tell me how to do so?

smathermather, 2016-03-25:
Thanks for looking into this, @titanous. @paulinus, would you be willing to look into this?

dakotabenjamin, 2016-05-16:
So I pulled OpenSfM from that commit onto ODM and it worked. See patch below

```
From cf71e60ee44b0f2620e4b97b658599059402f347 Mon Sep 17 00:00:00 2001
From: Dakota Benjamin <dakota.benjamin@gmail.com>
Date: Mon, 16 May 2016 15:55:14 -0400
Subject: [PATCH] Update OpenSfM to fix

---
 SuperBuild/cmake/External-OpenSfM.cmake | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/SuperBuild/cmake/External-OpenSfM.cmake b/SuperBuild/cmake/External
index fefd20c..33a1109 100644
--- a/SuperBuild/cmake/External-OpenSfM.cmake
+++ b/SuperBuild/cmake/External-OpenSfM.cmake
@@ -8,8 +8,8 @@ ExternalProject_Add(${_proj_name}
   STAMP_DIR         ${_SB_BINARY_DIR}/stamp
   #--Download step--------------
   DOWNLOAD_DIR      ${SB_DOWNLOAD_DIR}
-  URL               https://github.com/mapillary/OpenSfM/archive/a4b07056aec118
-  URL_MD5           42B2B1994C3309BBF4525C8CC1F6F741
+  URL               https://github.com/mapillary/OpenSfM/archive/21735be1776845
+#  URL_MD5           42B2B1994C3309BBF4525C8CC1F6F741
   #--Update/Patch step----------
   UPDATE_COMMAND    ""
   #--Configure step-------------
--
1.9.1

```

dakotabenjamin, 2016-05-17:
It looks like there's about 100 commits between our branch of OpenSfM and the master, and one commit that our branch has that may or may not have been incorporated into the master branch. That being said, I don't know from which commit in OpenSfM @paulinus will use when pushing SLAM to ODM. That may solve this issue outright.

closed 2018-03-23 by smathermather