OpenDroneMap/ODM #282: Clean install with opensfm problem

w0rcy, 2016-03-25:
I have a clean install of 14.04.4 Trusty that I have a clean install of ODM, that when I run, it goes through all of the photos a few times, then it gives the following error and stops:

Traceback (most recent call last):
  File "/home/rick/OpenDroneMap/src/OpenSfM/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/rick/OpenDroneMap/src/OpenSfM/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
b

quitting cause: 
	PYTHONPATH=/home/rick/OpenDroneMap/lib/python2.7/dist-packages "/home/rick/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

If you would like me to post logs, please let me know where they should be located in the normal user installation. If you would like for me to reinstall something, please let me know how to do that. I have been a unix/linux user for decades, but I am not a good administrator. Also, I have successfully used ODM, but it has been a while and on a different Linux install on the same computer.
Thanks

smathermather, 2016-03-25:
Hi @w0rcy -- what was your install process?

w0rcy, 2016-03-26:
I should have said fresh install of each. New hard drive, fresh Ubuntu 14.04.4, fresh ODM. This is the normal git clone and install as described in the wiki/1.-Installation page, word for word, but nothing more. No vagrant or docker, just straight Linux. It looks to me as if csfm is installed in the ODM directory, though it reports that it is not.

I have also tried the edgarriba method, but that didn't work for me, either, so I went back to the "main" branch.

dakotabenjamin, 2016-03-29:
Ok the installation logs for gh-pages (the main branch) are found in the logs/ directory. You'll probably want to look at the ceres logs first.

w0rcy, 2016-03-29:
The ceres_1_build.log doesn't have any errors or warnings, but here is the ceres_1_config.log, renamed [ceres_1_config.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/193913/ceres_1_config.txt) so that I can attach it, which might show something. I also include the 
[sysinfo.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/193918/sysinfo.txt) as more info.

dakotabenjamin, 2016-03-29:
Nope, no errors there. Possible errors and solutions:

1. csfm isn't in the PYTHONPATH. Check this with `echo $PYTHONPATH`
2. Opensfm didn't install correctly. Try deleting `opensfm/build` and `opensfm/cmake_build` before rebuilding using `python setup.py build`

w0rcy, 2016-03-30:
1. There was nothing in the main branch instructions about setting PYTHONPATH, only the edgarriba branch, which has paths that do not apply to the main branch, so I have none specified. What should the PYTHONPATH be? Should that be in the instructions?

2. I have to assume you mean ~/OpenDroneMap/src/OpenSfM/cmake_build. There is not a build directory in that path, though. I deleted that cmake_build and ran python setup.py build and in the output were a lot of this:

/home/rick/OpenDroneMap/src/OpenSfM/opensfm/src/third_party/vlfeat/vl/imopv_sse2.c:15:2: error: #error "Compiling with SSE2 enabled, but no __SSE2__ defined"
 #error "Compiling with SSE2 enabled, but no __SSE2__ defined"
  ^
In file included from /home/rick/OpenDroneMap/src/OpenSfM/opensfm/src/third_party/vlfeat/vl/imopv_sse2.c:22:0:
/usr/lib/gcc/i686-linux-gnu/4.8/include/emmintrin.h:31:3: error: #error "SSE2 instruction set not enabled"
 # error "SSE2 instruction set not enabled"
   ^

I hope this discussion is not just "fixing" my install, but to figure out what is not working with the normal installation process. If it would be better to work with the edgarriba branch, I can switch back to that.

Thanks

dakotabenjamin, 2016-03-30:
@paulinus may be better able to make sense of this...

I see one issue in mapillary/OpenSfM: https://github.com/mapillary/OpenSfM/issues/68

maybe try that?

dakotabenjamin, 2016-03-30:
If you're looking for a better install process, `dev` branch is where to go. It is what culminated from the edgarriba branch (and seems to be much more stable). 

I'm hoping to merge that once we figure out how to properly link the shared libraries and remove the need to set environment variables manually.

w0rcy, 2016-03-30:
Are there other dependencies I should install first?
Thanks

dakotabenjamin, 2016-03-30:
The dev [readme](https://github.com/OpenDroneMap/OpenDroneMap/blob/dev/README.md) has complete build instructions

w0rcy, 2016-03-30:
OK, I'm going through the dev flow. What does the line

bash configure.sh

do? I get:

bash: configure.sh: No such file or directory

Is that a file that should be in the odm directory, or should that be found
elsewhere? All I have done so far is the git and set the env variables.

w0rcy, 2016-03-30:
Oh, I see it in the dev branch on the web page. I need to get it to
download the proper branch, not what it says in the instructions.

w0rcy, 2016-03-30:
So the instructions should also have

git checkout dev

after the git command.

w0rcy, 2016-03-31:
I'll check back with you all in a few months and see how it is coming along. Please do a fresh install on a test machine now and then, just to see if it is relying on other things that you already have that aren't included in the git data.
Thanks

dakotabenjamin, 2016-04-04:
Sorry for the delayed response- it looks like you had a successful build, those errors in the make are just warnings. Let us know how it goes! :+1:

dakotabenjamin, 2016-05-16:
@w0rcy any update?

w0rcy, 2016-05-17:
I have not revisited this. I have been thinking, though, could my problem
be caused by my old AMD architecture causing something to not install
correctly? Maybe if I try Ubuntu 16.04LTS. I could not get 14.04, just
14.04.4. Have you tried it with this version?

dakotabenjamin, 2016-05-17:
I recently pushed an update to get 16.04 compatible. Another option, if you have any experience with [docker](https://hub.docker.com/r/opendronemap/opendronemap/) it will work in any architecture.

closed 2016-10-12 by dakotabenjamin