OpenDroneMap/ODM #285: No such file or directory: '/images/opensfm/reconstruction.json'

willcodeforfoo, 2016-04-06:
I apologize if this is a dumb question, but I receive the following error when trying to build one of the samples, and some of the examples are a bit confusing with the different branches:

```
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/mesh", line 23, in <module>
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 291, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/images/opensfm/reconstruction.json'
[ERROR]   quitting cause:
	PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/run_all /images/opensfm
returned with code 256.
```

I've built the `dev` branch of OpenDroneMap on a fresh EC2 instance with Docker and am attempting to run like so:

```
docker build -t opendronemap:latest .
docker run -it -v /home/ec2-user/odm_data/pacifica:/images opendronemap:latest
```

I cloned the sample project (`git clone --recursive git://github.com/OpenDroneMap/odm_data`), but did create an `images` subdirectory and moved the JPEGs there because I believe the `dev` branch requires them that way.

Full output is here, as well as a `tree` output of the `pacifica` samples directory after running: [Log at 2016-04-05 21.53.36.txt.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/205660/Log.at.2016-04-05.21.53.36.txt.zip)

Thanks!

dakotabenjamin, 2016-04-06:
Have you tried any other datasets yet?

dakotabenjamin, 2016-04-06:
I get this same issue on the pacifica set but not with others. I think the solution here is to remove pacifica from the odm_data repo. See: https://github.com/OpenDroneMap/odm_data/pull/4

dakotabenjamin, 2016-05-16:
Since this is likely a data problem, I'm going to close

closed 2016-05-16 by dakotabenjamin