OpenDroneMap/ODM #290: Crashing with Out of Memory

captbrando, 2016-04-21:
Folks:

Got a DJI Phantom 4. Updated the JSON focal length based on [this patch](https://github.com/OpenDroneMap/OpenDroneMap/pull/280/commits/df49d1131423c93296e4ca52c64ef82b2b650f46). All goes well, but now I'm getting a memory crash. Here's the last part of the console logging that lead to the out of memory crash:

```
---- Initial: 0 secs ----
Total pass fail0 fail1 refinepatch: 2788021 75178 2245954 466889 542067
Total pass fail0 fail1 refinepatch: 100 2.69646 80.5573 16.7463 19.4427
Expanding patches...
---- EXPANSION: 1307 secs ----
Total pass fail0 fail1 refinepatch: 6758471 5022679 476144 1259648 6282327
Total pass fail0 fail1 refinepatch: 100 74.3168 7.04514 18.6381 92.9549
FilterOutside
mainbody: 
Gain (ave/var): 0.651672 0.482354
5070366 -> 5044847 (99.4967%)   0 secs
Filter Exact: ****************************************************************************************************************************************************************************************************************************************************************************
5044847 -> 4976771 (98.6506%)   0 secs
FilterNeighbor: 4976771 -> 4768445 (95.814%)    0 secs
FilterGroups:   476
4768445 -> 4551972 (95.4603%)   0 secs
STATUS: 150731 0 6673328 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 
Expanding patches...
---- EXPANSION: 350 secs ----
Total pass fail0 fail1 refinepatch: 1182745 556999 162569 463177 1020176
Total pass fail0 fail1 refinepatch: 100 47.0938 13.7451 39.1612 86.2549
FilterOutside
mainbody: 
Gain (ave/var): 0.870642 0.553443
5108971 -> 5107944 (99.9799%)   0 secs
Filter Exact: ****************************************************************************************************************************************************************************************************************************************************************************
5107944 -> 5085331 (99.5573%)   0 secs
FilterNeighbor: 5085331 -> 5023647 (98.787%)    0 secs
FilterGroups:   502
5023647 -> 4941768 (98.3701%)   0 secs
STATUS: 180331 0 7663866 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 
Expanding patches...
---- EXPANSION: 307 secs ----
Total pass fail0 fail1 refinepatch: 718548 333420 110555 274573 607993
Total pass fail0 fail1 refinepatch: 100 46.4019 15.3859 38.2122 84.6141
FilterOutside
mainbody: 
Gain (ave/var): 1.0802 0.623972
5275188 -> 5274499 (99.9869%)   0 secs
Filter Exact: ***************Killed
b

quitting cause: 
       "/opt/OpenDroneMap/bin/pmvs2" pmvs/ option-0000
returned with code 35072.

```

Now, this is a rather large stitch. 700 images. There has to be some kind of inefficient handling of memory somewhere. I did end up running this on the MapsMadeEasy website, and it generated all my files just fine. The end TIF was 2G, the JPG was 200 megs.

What else can I include on this to help debug the problem?

Stats of the VM: Ubuntu 14.04. 44Gig of HD, 20G RAM.

dakotabenjamin, 2016-04-25:
@captbrando what's your CPU/ number of cores assigned to the VM?

@smathermather do you recall some weird stuff happening when there's a lot more RAM than cpu cores?

smathermather, 2016-04-25:
Yup, the ratio of CPUs to memory has to be right, but I don't think we ever established that threshold.

captbrando, 2016-04-25:
@dakotabenjamin 16 cores were allocated, and 20 gigs of ram. We kept upping it to see if it was a memory issue. 

Do we need to keep things equal? 4 cores, 4 gigs?

dakotabenjamin, 2016-04-25:
I usually aim for that. You could also try resizing the images to be smaller as well.

captbrando, 2016-04-28:
OK, it crashed again. What logs can I provide to help tech the problem?

dakotabenjamin, 2016-05-02:
For this problem in particular, I don't think there are any logs that will help. 

However, I think there's a few things you ought to try:

use `--cmvs-maxImages 250`. The default is 500, but reducing the number of images per cmvs cluster will reduce memory usage. You can keep trying to reduce that number until the cmvs process gets stuck (infinite loop) to maximize this. 

If that doesn't work, change the thread count for pmvs
  1. edit pmvs/recon0/option-0000 such that `n` in the line `CPU n` is smaller. Then rerun odm using `--start-with pmvs` so that it doesn't rewrite the options file.

dakotabenjamin, 2016-05-16:
@captbrando any luck?

captbrando, 2016-05-16:
I ran a quick test last week and it still crashed, but I ended up dropping it for the time being. Are there any specific debugging logs or things I can run in the background for the next attempt?

dakotabenjamin, 2016-05-16:
Are you willing to share the dataset? We can set up a private transfer and debug on our end. @smathermather

captbrando, 2016-05-16:
Absolutely! It’s 700 shots of some farmland, so just let me know and I’ll upload!

smathermather, 2018-03-23:
Should be resolved. Please reopen if otherwise.

closed 2018-03-23 by smathermather