OpenDroneMap/ODM #292: Install in Ubuntu 16.04

OscarLS91, 2016-04-25:
Trying to run install.sh, and the script doesn't end in "script finished". Log file says that couldn't find gFortran

dakotabenjamin, 2016-04-25:
Thanks for reminding me that a new LTS is out! 

I'm having trouble getting a Xenial vagrant machine up and running, but once I do I'll try to debug and update.

terramars, 2016-05-04:
I got past the gFortran issue (that's trivial just change to lower case / maybe needs a symbolic link), there are problems with ceres and glog.  I got it to work (with ceres or without ceres), but getting it to work with ceres required building it with MINIGLOG as an option, which you're not supposed to do.

Here's the error message fwiw : 

g++ -o bundler -O3 -Wall -Wno-unused-result -std=gnu++0x -I../lib/imagelib -I../lib/sfm-driver -I../lib/matrix -I../lib/5point -I../lib/sba-1.5 -I../lib/ann_1.1_char/include -I../include -I../../../include -I/usr/include/eigen3 -L../lib -L../lib/ann_1.1_char/lib -L../../../lib \
	-D__NO_UI__ -D__BUNDLER__ -D__BUNDLER_DISTR__ -D__USE_CERES__ BaseApp.o BundlerApp.o keys.o Register.o Epipolar.o Bundle.o BundleFast.o MatchTracks.o Camera.o Geometry.o ImageData.o SifterUtil.o BaseGeometry.o BundlerGeometry.o BoundingBox.o BundleAdd.o ComputeTracks.o BruteForceSearch.o BundleIO.o ProcessBundle.o BundleTwo.o Decompose.o RelativePose.o Distortion.o TwoFrameModel.o LoadJPEG.o BundleCeres.o -lceres -lcholmod -lccolamd -lcolamd -lamd -lcamd -lcxsparse -lsuitesparseconfig -lgomp -lglog -lpthread -limage -lsfmdrv -lsba.v1.5 -lmatrix -lz -llapack -lblas -lcblas -lminpack -lm -l5point -ljpeg -lANN_char -lgfortran
BundleCeres.o: In function `BundlerApp::RunSFM_Ceres(int, int, int, bool, camera_params_t*, v3_t*, int*, v3_t*, std::vector<std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >, std::allocator<std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > > >&, int, int, int, double, double*, double*, double*, double*, bool, bool, bool)':
BundleCeres.cpp:(.text+0x12d9): undefined reference to `ceres::Solver::Summary::FullReport() const'
BundleCeres.o: In function `std::string* google::MakeCheckOpString<int, int>(int const&, int const&, char const*)':
BundleCeres.cpp:(.text._ZN6google17MakeCheckOpStringIiiEEPSsRKT_RKT0_PKc[_ZN6google17MakeCheckOpStringIiiEEPSsRKT_RKT0_PKc]+0x49): undefined reference to `google::base::CheckOpMessageBuilder::NewString()'
BundleCeres.o: In function `std::string* google::MakeCheckOpString<int, ceres::DimensionType>(int const&, ceres::DimensionType const&, char const*)':
BundleCeres.cpp:(.text._ZN6google17MakeCheckOpStringIiN5ceres13DimensionTypeEEEPSsRKT_RKT0_PKc[_ZN6google17MakeCheckOpStringIiN5ceres13DimensionTypeEEEPSsRKT_RKT0_PKc]+0x49): undefined reference to `google::base::CheckOpMessageBuilder::NewString()'
collect2: error: ld returned 1 exit status
Makefile:73: recipe for target 'bundler' failed
make: *** [bundler] Error 1

terramars, 2016-05-04:
A major source of issues (unrelated to the Ceres problem I "think") is that most of the libraries are in /usr/lib/x86_64-linux-gnu/ now, and the linker is not able to find them.  I am creating symlinks as I go, these ones are trivial to debug though.  Just go through and when shit fails create the symlink.  I'm sure you could also update the script to just know about the folder entirely.

deltafactory, 2016-05-06:
I fixed the gFortran manually and stopped on
> extract_utm
    - configuring odm_extract_utm
    - building odm_extract_utm

dakotabenjamin, 2016-05-11:
So the symlink would be from /usr/lib to /usr/lib/x86_64-linux-gnu/ ?

deltafactory, 2016-05-11:
Yes.

dakotabenjamin, 2016-05-11:
Just pushed a branch `xenial-compatibility`. Please test and report. Instead of directly linking the libproj.so library, I used find_library() to make cmake more os-agnostic.

dakotabenjamin, 2016-05-11:
https://github.com/OpenDroneMap/OpenDroneMap/pull/299

deltafactory, 2016-05-12:
Looks like it work after git pull/checkout, commenting out .bashrc entries and opening new shell.

closed 2016-05-16 by dakotabenjamin