OpenDroneMap/ODM #293: Install doesn't print "script finished" on Ubuntu 14.04

DarylWM, 2016-04-27:
I followed the instructions [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/1.-Installation), but it didn't get very far.

installing required packages
    - updating

[sysinfo.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/238182/sysinfo.txt)
[apt-get_get.log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/238185/apt-get_get.log.txt)

It seems to be breaking on the sudo apt-get update.

DarylWM, 2016-04-27:
Found it wasn't a problem with ODM; it was my apt-get sources.list. Fixed the update problem by replacing the contents of `/etc/apt/sources.list` with:

```
deb mirror://mirrors.ubuntu.com/mirrors.txt trusty main restricted universe multiverse
deb mirror://mirrors.ubuntu.com/mirrors.txt trusty-updates main restricted universe multiverse
deb mirror://mirrors.ubuntu.com/mirrors.txt trusty-backports main restricted universe multiverse
deb mirror://mirrors.ubuntu.com/mirrors.txt trusty-security main restricted universe multiverse
```
and doing a `sudo apt-get update`, then running `./install.sh` again.

closed 2016-04-27 by DarylWM

smathermather, 2016-04-28:
Excellent. Thanks for the update @DarylWM.