OpenDroneMap/ODM #295: question about the versions

xialang2012, 2016-05-05:
Hi, I found some different new branches belong to this project since the last time I fork it. I am not exactly know the different means between them ,such as Mvs-Texturing, update-mvstex. Besides, I have ever found one branch which can blending the texturing and ortho result, but now I can not find it. Has it been removed from the project? Thank you.

deltafactory, 2016-05-16:
I don't know about the branches you mention but there was major restructuring: `gh-pages` now only contains github web content as it should. `master` contains the current release. `dev` contains an upcoming release and is looking pretty stable but significantly changes the build and run process including output directory structure. The Readme.md for the `dev` branch details much of that.

Other branches were probably development modules that have been merged into dev and deleted (though I still see `mvs-texturing-addition` at the moment.) Hope that helps.

dakotabenjamin, 2016-05-16:
@deltafactory is correct, I hope the restructuring has cleared things up a bit. The `mvs-texturing-addition` is a [feature or topic branch](https://git-scm.com/book/en/v2/Git-Branching-Branching-Workflows#Topic-Branches) that needs some work still (needs a rebase and some debugging) before it can be merged into the `dev` development branch.

xialang2012, 2016-05-17:
@deltafactory thank you for you detailed reply.
@dakotabenjamin I have already test the mvs-texturing-addition branch and it worked well for me. It seems that the mvs branches provided blending algorithm that can provide a uniform color texture image.

closed 2017-01-19 by dakotabenjamin