OpenDroneMap/ODM #298: Missing odm_georef options in scripts/odm_georeferencing.py

wildintellect, 2016-05-07:
scripts/odm_georeferencing.py is missing some options when calling odm_georef for use with a gcp text file. ```python run.py --odm_georeferencing-useGcp --odm_georeferencing-gcpFile gcp_list.txt```

Missing options are, which odm_georef says are required:
-imagesPath {imgs}
-imagesListPath {imgs_list}

Near line 98 currently
```
 if self.params.use_gcp and \
               io.file_exists(tree.odm_georeferencing_coords):

                system.run('{bin}/odm_georef -bundleFile {bundle} -inputCoordFile {coords} '
                           '-bundleResizedTo {size} -inputFile {model} -outputFile {model_geo} '
                           '-inputPointCloudFile {pc} -outputPointCloudFile {pc_geo} '
                           '-logFile {log} -georefFileOutputPath {geo_sys} -gcpFile {gcp} '
                           '-outputCoordFile {coords}'.format(**kwargs))
```

Example proper command from Stephen
```
~/OpenDroneMap/build/bin/odm_georef \
-bundleFile ~/anna/opensfm/bundle_r000.out \
-gcpFile ~/anna/gcp_list.txt \
-imagesPath ~/anna/images_resize/ \
-imagesListPath ~/anna/opensfm/list_r000.out \
-bundleResizedTo 2400 \
-inputFile ~/anna/odm_texturing/odm_textured_model.obj \
-outputFile ~/anna/odm_texturing/odm_textured_model_geo.obj \
-outputCoordFile /odm_data/copr/odm_georeferencing/coords.txt \
-inputPointCloudFile ~/anna/pmvs/recon0/models/option-0000.ply \
-outputPointCloudFile ~/anna/odm_georeferencing/odm_georeferenced_model.ply \
-logFile ~/anna/odm_georeferencing/odm_georeferencing_log_custom.txt \
-georefFileOutputPath ~/anna/odm_georeferencing/odm_georeferencing_model_geo.txt
```

wildintellect, 2016-05-09:
I'm thinking that section of code should probably look more like the following. It doesn't make sense to ask if the user wants to use a gcp file and then check for the EXIF based coords file.
```
 if self.params.use_gcp and \
               io.file_exists(self.params.gcp_file):

                system.run('{bin}/odm_georef -bundleFile {bundle} -bundleResizedTo {size} '
                           '-inputFile {model} -outputFile {model_geo} '
                           '-inputPointCloudFile {pc} -outputPointCloudFile {pc_geo} '
                           '-imagesPath {imgs} -imagesListPath {imgs_list} -gcpFile {gcp} '
                           '-logFile {log} -georefFileOutputPath {geo_sys} '
                           '-outputCoordFile {coords}'.format(**kwargs))
            else:
```

dakotabenjamin, 2016-05-09:
I thought I fixed that.... whoops

closed 2018-03-28 by dakotabenjamin