OpenDroneMap/ODM #301: Can not recognize images directory on the opendronemap/opendronemap:dev Docker image

jangbi882, 2016-05-13:
I ran the opendronemap/opendronemap:latest Docker image successfully.
But, can't run opendronemap/opendronemap:dev Docker image because it can't recognize images folder.
Do you know why?

I currently  use Docker for Windows beta, it run based on Hyper-V.
[image]

Below is my logs.
```
C:\Users\jangbi>docker run -v c:/Users/jangbi/odm_sample:/images opendronemap/opendronemap:dev
[INFO]    Initializing OpenDroneMap app - Fri May 13 06:38:45  2016
[INFO]    Running ODM Load Dataset Cell
[ERROR]   You must put your pictures into an <images> directory
[INFO]    OpenDroneMap app finished - Fri May 13 06:38:46  2016

C:\Users\jangbi>docker run -v c:/Users/jangbi/odm_sample:/images opendronemap/opendronemap:latest

  - configuration:
     start_with : resize
     end_with : odm_orthophoto
     force_focal : None
     odm_texturing_textureWithSize : 3600
     matcher_neighbors : 8
     matcher_threshold : 2.0
     cmvs_maxImages : 500
     odm_orthophoto_resolution : 20.0
     pmvs_threshold : 0.7
     zip_results : False
     resize_to : 2400
     use_opensfm : True
     min_num_features : 6000
     odm_meshing_octreeDepth : 9
     odm_meshing_solverDivide : 9
     pmvs_wsize : 7
     odm_texturing_textureResolution : 4096
     matcher_ratio : 0.6
     odm_meshing_samplesPerNode : 1.0
     odm_meshing_maxVertexCount : 100000
     matcher_distance : 0
     pmvs_level : 1
     run_only : None
     job_dir_name : None
     odm_georeferencing_useGcp : False
     odm_georeferencing_gcpFile : gcp_list.txt
     pmvs_csize : 2
     pmvs_minImageNum : 3
     force_ccd : None

  - source files - Fri May 13 06:44:07  2016
     using img_4851.jpg     dimensions: 4608x3456 / focal: 5.0mm / ccd: 6.58mm
     using img_4852.jpg     dimensions: 4608x3456 / focal: 5.0mm / ccd: 6.58mm
     using img_4853.jpg     dimensions: 4608x3456 / focal: 5.0mm / ccd: 6.58mm
     ....
```

dakotabenjamin, 2016-05-13:
In the dev version we've updated how projects are handled in the file system. Your project folder should be set up like this:
```
|-- /path/to/project/
    |-- images/
        |-- img-1234.jpg
        |-- ...
```

in Docker, you will specify `/path/to/project/` as your `/images` volume, not `/path/to/project/images`. I see now how this can be a point of confusion, so maybe it should be improved.

dakotabenjamin, 2016-05-13:
To be clear, you simply need to place your .jpg files in a folder called "images" inside their current folder. 
```
mkdir c:/Users/jangbi/odm_sample/images
mv  c:/Users/jangbi/odm_sample/*.jpg c:/Users/jangbi/odm_sample/images
```

jangbi882, 2016-05-14:
Thank you.
I can run the dev docker with your advice.
But the dev docker looks not work properly.

I attach running log and list of result files.

[odm_dev_docker_log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/264345/odm_dev_docker_log.txt)
[odm_result_files.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/264344/odm_result_files.txt)

dakotabenjamin, 2016-05-16:
I'm a little confused because the log sort of disappears at OpenSfM but it creates a pmvs point cloud (./pmvs/recon0/models/options-0000.ply). This was Boruszyn dataset, right?

dakotabenjamin, 2016-05-16:
To see if it's something related to the client, I've requested access to the Docker for Windows beta, but my main machine is Windows 7 so I'll have to find one with Windows 10 Pro/Enterprise to test it on @smathermather

dakotabenjamin, 2016-05-16:
Found the issue. Forgot the put the compile directive in the Dockerfile. 

Will fix and push later today. 

If you wanna clone locally and test it yourself, change lines 69-72 to look like this:
```
RUN bash ./configure.sh && \
    mkdir build && cd build && cmake .. && make && cd .. && \
    chown -R odm:odm /code
USER odm
```

jangbi882, 2016-05-17:
The new Docker image is working well, odm_meshing and odm_texturing are created.
But still not create a orthoimage.

[image]

My log and created files list are attached this post.
[odm_docker_log_20160517.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/267316/odm_docker_log_20160517.txt)
[odm_result_files_20160517.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/267317/odm_result_files_20160517.txt)

dakotabenjamin, 2016-05-17:
try this:
```
docker run -v c:/Users/jangbi/odm_sample:/images opendronemap/opendronemap:dev --odm_georeferencing-useGcp --rerun-from odm_georeferencing
```

jangbi882, 2016-05-17:
A error accrued but some files is create in odm_georeferencing folder.
[image]

Log messages
'''
C:\Users\jangbi>docker run -v c:/Users/jangbi/odm_sample:/images opendronemap/opendronemap:dev --odm_georeferencing-useGcp --rerun-from odm_georeferencing
[INFO]    Initializing OpenDroneMap app - Tue May 17 12:37:25  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /images/images_resize
[DEBUG]   Loaded img_4851.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4852.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4853.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4854.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4858.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4871.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4873.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4880.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4881.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4883.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4885.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4890.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4908.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4909.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4910.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4911.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4912.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4915.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4918.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4925.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4926.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4928.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4929.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4930.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4933.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4934.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4935.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4937.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4939.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4940.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4946.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4949.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4950.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4951.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4953.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4955.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4961.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4964.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4965.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4966.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4967.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4969.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[DEBUG]   Loaded img_4970.jpg | camera: canon powershot a3300 is | dimensions: 2400 x 1800 | focal: 5.0 | ccd: 6.16
[... cut ...]
[ERROR]   Could not parse coordinates. Bad CRS supplied: PUWG1992

Traceback (most recent call last):
  File "/code/run.py", line 38, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 118, in process
    geo_ref.utm_to_latlon(tree.odm_georeferencing_latlon, photo, idx)
  File "/code/opendm/types.py", line 209, in utm_to_latlon
    gcp = self.gcps[idx]
IndexError: list index out of range
'''
[odm_files_20160517_2.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/268172/odm_files_20160517_2.txt)

dakotabenjamin, 2016-05-17:
You need an updated version of the gcp_file. I realized that I never pushed the changes to the repository, so you'll need to pull the updated changes from here: https://github.com/merkato/odm_boruszyn_kap/tree/01a4f5cf0838a0fb0bc33ce16555947a76555f0f

dakotabenjamin, 2016-05-17:
or just change the first line of gcp_list.txt to "EPSG:2180"

jangbi882, 2016-05-17:
My gcp_list.txt is not on images folder, its on project root folder.
I will move this file to images folder and retry.

dakotabenjamin, 2016-05-17:
keep it in the project root

jangbi882, 2016-05-17:
Finally I got it!
Thank you so so so much.

[image]

dakotabenjamin, 2016-05-17:
Glad to have been of help :smile:

closed 2016-05-17 by dakotabenjamin