OpenDroneMap/ODM #306: Using GCPs with angular coordinates (i.e WGS84) bug

dakotabenjamin, 2016-05-16:
when using GCP files with WGS84 (latlon), the model will scale and become very rough. I'll post a comparison. Basically when going from hundreds of pixels or whatever in the model to a fraction of a degree in the angular coodinate system, a lot of data seems to be thrown out. When using EXIF data on the photos to georeference, the program converts that to the nearest UTM, I believe. The same should be done for when using GCP files. 

[image]

yjmenezes, 2017-06-10:
@dakotabenjamin 

I think gcp_list.txt could be projected from Lat/Lon to UTM as a preparation task.
it could be done with Proj4 using shell scripts. ( under /contrib/ ? )
That is how i do.  This way, ODM code do not need to care about Datum, units, projection.

dakotabenjamin, 2017-06-12:
It probably should happen internally though. @fredlllll when perusing through GeoRef.cpp did you happen across this?

fredlllll, 2017-06-12:
wasnt really looking over all of it, but i have the feeling that this here could be the culprit? https://github.com/OpenDroneMap/OpenDroneMap/blob/66092078dce699fca1837698c76a45a62977f994/modules/odm_georef/src/Georef.cpp#L820

also some of these calculations are done using floats, that could give you precision problems

cast to floats https://github.com/OpenDroneMap/OpenDroneMap/blob/66092078dce699fca1837698c76a45a62977f994/modules/odm_georef/src/Georef.cpp#L804

pierotofy, 2019-06-21:
This has been fixed in the `speedimprov` branch.

closed 2019-06-21 by pierotofy