OpenDroneMap/ODM #307: Suggestion about obtain the orthophoto

xialang2012, 2016-05-18:
Now, the ODM products  an orthophoto image by using the following logic:
`Mesh-->Texturing-->Georef(ConvertUTM)-->Orthophoto`
But, some images have no geolocation information, and we will be block in this step if we wish to get the orthophoto. 
May be we can ignore the  Georef(ConvertUTM) step, but usually we will get a bad orthophoto (for me, I have tried many times). So is it possible for us to get the orthophoto without geolocation information on the images? Thanks.

dakotabenjamin, 2016-05-18:
`dev` branch will create a .png orthophoto regardless of whether the photos are georeferenced.

dakotabenjamin, 2016-05-18:
Actually that is not true, although it used to be the case. I think it is a good idea, and is possible.

dakotabenjamin, 2016-05-18:
Here's what happens now:
```
[INFO]    Running ODM Georeferencing Cell
[WARNING] Warning: No coordinates file. Generating coordinates file in: /vagrant_data/odm_data/boruszyn_kap/odm_georeferencing/coords.txt
[DEBUG]   running /home/vagrant/opendronemap/build/bin/odm_extract_utm -imagesPath /vagrant_data/odm_data/boruszyn_kap/images/ -imageListFile /vagrant_data/odm_data/boruszyn_kap/opensfm/list_r000.out -outputCoordFile /vagrant_data/odm_data/boruszyn_kap/odm_georeferencing/coords.txt -logFile /vagrant_data/odm_data/boruszyn_kap/odm_georeferencing/odm_georeferencing_utm_log.txt
Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.
[ERROR]   quitting cause:
        /home/vagrant/opendronemap/build/bin/odm_extract_utm -imagesPath /vagrant_data/odm_data/boruszyn_kap/images/ -imageListFile /vagrant_data/odm_data/boruszyn_kap/opensfm/list_r000.out -outputCoordFile /vagrant_data/odm_data/boruszyn_kap/odm_georeferencing/coords.txt -logFile /vagrant_data/odm_data/boruszyn_kap/odm_georeferencing/odm_georeferencing_utm_log.txt
returned with code 256.

An error occurred. Check stdout above or the logs.
```

In the future, we could catch that error, throw up a warning that we couldn't georeference, then continue with flag for orthophoto to only produced the unreferenced image.

xialang2012, 2016-06-14:
Great, I am waiting for the update, Thank you, @dakotabenjamin

AminSarafraz, 2016-07-07:
Hi @dakotabenjamin, I have the same problem. I do not want to georeference the image, I just need the orthophoto. I will do the georeferencing later. I tried to find a way to ignore the georeferencing step, but I was not able to find a solution. The "--run-only" option that is suggested [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/3.-Run-Time-Parameters) is not available in the latest version that I checked out. Would you please tell me how I can accomplish this? 

I am running the OpenDroneMap on Centos 7 through docker and I am using Langley data for testing. Here is the error that I get:

[ERROR]   quitting cause: 
	/code/build/bin/odm_extract_utm -imagesPath /images/images/ -imageListFile /images/opensfm/list_r000.out -outputCoordFile /images/odm_georeferencing/coords.txt -logFile /images/odm_georeferencing/odm_georeferencing_utm_log.txt
returned with code 256.

closed 2020-04-14 by smathermather