OpenDroneMap/ODM #308: Installation failed, seemed the configuration of odm_texturing_1_build was not successful

tianyufang1958, 2016-05-18:
This the log file and the configuring incomplete. Not sure what the problem is, the cuda version is lower, but is it mandatory to have CUDA?

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Could NOT find openni (missing:  OPENNI_INCLUDE_DIRS) 
** WARNING ** io features related to openni will be disabled
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- looking for PCL_COMMON
-- Found PCL_COMMON: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_common.so  
-- looking for PCL_OCTREE
-- Found PCL_OCTREE: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_octree.so  
-- looking for PCL_IO
-- Found PCL_IO: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_io.so  
-- looking for PCL_KDTREE
-- Found PCL_KDTREE: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_kdtree.so  
-- looking for PCL_SEARCH
-- Found PCL_SEARCH: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_search.so  
-- looking for PCL_SURFACE
-- Found PCL_SURFACE: /home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_surface.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_common.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_common.so;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_octree.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_octree.so;/usr/lib/libOpenNI2.so;vtkIOAMR;vtkFiltersAMR;vtkFiltersGeneral;vtkCommonComputationalGeometry;vtkCommonDataModel;vtkCommonMath;vtkCommonCore;vtksys;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkFiltersCore;vtkCommonExecutionModel;vtkParallelCore;vtkIOLegacy;vtkIOCore;vtkzlib;vtkhdf5_hl;vtkhdf5;vtkGeovisCore;vtkIOXML;vtkIOGeometry;vtkjsoncpp;vtkIOXMLParser;vtkexpat;vtkInfovisLayout;vtkFiltersModeling;vtkFiltersSources;vtkImagingHybrid;vtkIOImage;vtkDICOMParser;vtkmetaio;vtkjpeg;vtkpng;vtktiff;vtkImagingCore;vtkInfovisCore;vtkFiltersExtraction;vtkFiltersStatistics;vtkImagingFourier;vtkalglib;vtkInteractionStyle;vtkRenderingCore;vtkCommonColor;vtkFiltersGeometry;vtkInteractionWidgets;vtkFiltersHybrid;vtkImagingSources;vtkImagingGeneral;vtkRenderingAnnotation;vtkImagingColor;vtkRenderingFreeType;vtkfreetype;vtkftgl;vtkRenderingVolume;vtkViewsCore;vtkproj4;vtkFiltersHyperTree;vtkgl2ps;vtkIOExport;vtkRenderingContext2D;vtkRenderingGL2PS;vtkRenderingContextOpenGL;vtkRenderingOpenGL;vtkRenderingLabel;vtkRenderingImage;vtkIOLSDyna;vtkInteractionImage;vtkIOParallelXML;vtkViewsContext2D;vtkRenderingFreeTypeOpenGL;vtkRenderingVolumeOpenGL;vtkImagingMath;vtkFiltersParallelImaging;vtkFiltersImaging;vtkFiltersParallel;vtkIOMovie;vtkoggtheora;vtkChartsCore;vtkIOPLY;vtkIOVideo;vtkIOExodus;vtkexoIIc;vtkNetCDF;vtkNetCDF_cxx;vtkFiltersSMP;vtklibxml2;vtkDomainsChemistry;vtkIOEnSight;vtkImagingMorphological;vtkIONetCDF;vtkFiltersProgrammable;vtkIOSQL;vtksqlite;vtkFiltersSelection;verdict;vtkRenderingLIC;vtkIOParallel;vtkIOMINC;vtkViewsInfovis;vtkFiltersTexture;vtkImagingStatistics;vtkImagingStencil;vtkRenderingLOD;vtkFiltersFlowPaths;vtkFiltersGeneric;vtkFiltersVerdict;vtkIOInfovis;vtkIOImport;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_io.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_io.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_kdtree.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_kdtree.so;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_search.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_search.so;optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so;optimized;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_surface.so;debug;/home/libo/Downloads/OpenDroneMap/lib/pcl/lib/libpcl_surface.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so;/usr/lib/libOpenNI2.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;vtkIOAMR;vtkFiltersAMR;vtkFiltersGeneral;vtkCommonComputationalGeometry;vtkCommonDataModel;vtkCommonMath;vtkCommonCore;vtksys;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkFiltersCore;vtkCommonExecutionModel;vtkParallelCore;vtkIOLegacy;vtkIOCore;vtkzlib;vtkhdf5_hl;vtkhdf5;vtkGeovisCore;vtkIOXML;vtkIOGeometry;vtkjsoncpp;vtkIOXMLParser;vtkexpat;vtkInfovisLayout;vtkFiltersModeling;vtkFiltersSources;vtkImagingHybrid;vtkIOImage;vtkDICOMParser;vtkmetaio;vtkjpeg;vtkpng;vtktiff;vtkImagingCore;vtkInfovisCore;vtkFiltersExtraction;vtkFiltersStatistics;vtkImagingFourier;vtkalglib;vtkInteractionStyle;vtkRenderingCore;vtkCommonColor;vtkFiltersGeometry;vtkInteractionWidgets;vtkFiltersHybrid;vtkImagingSources;vtkImagingGeneral;vtkRenderingAnnotation;vtkImagingColor;vtkRenderingFreeType;vtkfreetype;vtkftgl;vtkRenderingVolume;vtkViewsCore;vtkproj4;vtkFiltersHyperTree;vtkgl2ps;vtkIOExport;vtkRenderingContext2D;vtkRenderingGL2PS;vtkRenderingContextOpenGL;vtkRenderingOpenGL;vtkRenderingLabel;vtkRenderingImage;vtkIOLSDyna;vtkInteractionImage;vtkIOParallelXML;vtkViewsContext2D;vtkRenderingFreeTypeOpenGL;vtkRenderingVolumeOpenGL;vtkImagingMath;vtkFiltersParallelImaging;vtkFiltersImaging;vtkFiltersParallel;vtkIOMovie;vtkoggtheora;vtkChartsCore;vtkIOPLY;vtkIOVideo;vtkIOExodus;vtkexoIIc;vtkNetCDF;vtkNetCDF_cxx;vtkFiltersSMP;vtklibxml2;vtkDomainsChemistry;vtkIOEnSight;vtkImagingMorphological;vtkIONetCDF;vtkFiltersProgrammable;vtkIOSQL;vtksqlite;vtkFiltersSelection;verdict;vtkRenderingLIC;vtkIOParallel;vtkIOMINC;vtkViewsInfovis;vtkFiltersTexture;vtkImagingStatistics;vtkImagingStencil;vtkRenderingLOD;vtkFiltersFlowPaths;vtkFiltersGeneric;vtkFiltersVerdict;vtkIOInfovis;vtkIOImport (Required is at least version "1.7") 
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find CUDA: Found unsuitable version "5.5", but required is exact
  version "7.5" (found /usr)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:313 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindCUDA.cmake:806 (find_package_handle_standard_args)
  /usr/local/share/OpenCV/OpenCVConfig.cmake:219 (find_package)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/libo/Downloads/OpenDroneMap/src/odm_texturing/CMakeFiles/CMakeOutput.log".

tianyufang1958, 2016-05-18:
The cuda problem seemed solved after upgrading to cuda7.5 but new problem occured for qt

Scanning dependencies of target odm_texturing
[ 25%] [ 50%] [ 75%] Building CXX object CMakeFiles/odm_texturing.dir/src/Logger.cpp.o
[100%] Building CXX object CMakeFiles/odm_texturing.dir/src/modifiedPclFunctions.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/main.cpp.o
Linking CXX executable odm_texturing
/usr/bin/ld: warning: libicui18n.so.54, needed by //opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libicuuc.so.54, needed by //opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libicudata.so.54, needed by //opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5, not found (try using -rpath or -rpath-link)
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_clone_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `u_strToLower_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_setMillis_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_fromUnicode_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_getDefaultName_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_inDaylightTime_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_get_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_close_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_openTimeZones_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_toUnicode_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `u_strToUpper_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_close_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_getMaxCharSize_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_countAvailable_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_getDefaultTimeZone_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `uenum_next_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_getStandardName_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucol_setAttribute_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucol_strcoll_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_setSubstChars_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_getTimeZoneDisplayName_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_openCountryTimeZones_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_open_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucol_open_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucol_close_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucol_getSortKey_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_getAvailableName_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_getDSTSavings_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_openTimeZoneIDEnumeration_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucal_open_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `u_errorName_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `uenum_close_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_getAlias_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_compareNames_54'
//opt/Qt/5.5/gcc_64/lib/libQt5Core.so.5: undefined reference to `ucnv_countAliases_54'
collect2: error: ld returned 1 exit status
make[2]: *** [odm_texturing] Error 1
make[1]: *** [CMakeFiles/odm_texturing.dir/all] Error 2
make: *** [all] Error 2

tianyufang1958, 2016-05-18:
But I can find the lib file in //opt/Qt/5.5/gcc_64/lib/

tianyufang1958, 2016-05-18:
Sorry I just keep posting, but problems solved once I install the newest version of ICU.

dakotabenjamin, 2016-05-18:
hmm, good. I've never seen these issues before. Can you tell us your architecture?

tianyufang1958, 2016-05-18:
Hi, my PC is intel Xeno CPU X5560, graphic card Quadro K2200, running on Ubuntu 14.04 64bit.

One more question. Our project is to orthomosaick tiff images from drone with gps coordinates data in separate txt file. Could you please let me know if opendronemap can do this?

dakotabenjamin, 2016-05-18:
Without knowing the format of the text file, I don't know. You can do one of two things. 
1. There are python scripts or other programs for geotagging photos from .gpx files. You might be able to manipulate those scripts to your file format. I use [mapillary's script](https://github.com/mapillary/mapillary_tools/blob/master/python/geotag_from_gpx.py)
2. You can create a gcp_list.txt file from ground control points in your images. see [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/2.-Running-OpenDroneMap#running-odm-with-ground-control) for more info on that. The GCPs only need to be very recognizable landmarks in multiple images.

dakotabenjamin, 2016-05-18:
ODM assumes 3 band imagery, so whatever is input into the process should be as such. You will have to manually convert which ever column from that text file into lat-long coordinates and write to the EXIF of the images. There are tools for writing EXIF data such as pyexiv2 of [gexiv2](https://wiki.gnome.org/Projects/gexiv2).

dakotabenjamin, 2016-06-15:
@tianyufang1958 has this issue been resolved?

tianyufang1958, 2016-06-16:
Hi Dakota,

Finally I converted the 32bit float tiff to 16 bit unsigned int and mosaic with Agisoft photo with importing the gps coordinates as a separate .txt file. Could you please let me know if OpenDroneMap can use separate txt file for mosaic? or it has be stored in the header file of image?

Regards

Dr. Bo LI
Image analysis specialist & Bioinformatician | Genetics and Crop Improvement
NIAB EMR | New Road, East Malling, Kent, ME19 6BJ
01732 523734   bo.li@emr.ac.uk

This email and any files transmitted with it may contain confidential or privileged information which are intended for the addressee only. Access to this email by anyone else is unauthorised and any disclosure, copying, distribution or any action taken or omitted to be taken or reliance on it, is prohibited and may be unlawful. If you are not the named addressee or the person responsible for delivering the message to the named addressee, please delete the email and any attachments and contact: enquiries:emr.ac.uk For any information about NIAB EMR or for a copy of the Terms of Business, please contact +44 (0)1732 843833 or visit www.emr.ac.uk. NIAB EMR is a charitable company limited by guarantee | Registered in England : 09894859 | Registered Charity : 1165055. Registered office: NIAB EMR, Huntingdon Road, Cambridge, Cambridgeshire, United Kingdom, CB3 0LE, UK.

dakotabenjamin, 2016-06-16:
What does the text file look like? Right now you need to either have coords in the exif data or a [GCP file](https://github.com/OpenDroneMap/OpenDroneMap/wiki/2.-Running-OpenDroneMap#running-odm-with-ground-control). However, if we can determine a standard text format for coords in a text file, for example:
```
IMG_0001.jpg, -81.2342, 45.1233, 234.2
...
```
then it would be pretty simple to incorporate into the workflow.

tianyufang1958, 2016-06-16:
Hi Dakota,

Below is the format of the txt file

#label    X/East    Y/North    Altitude
Calib_K00000.tif    0.395486666667    52.6995133333    -2.0
Calib_K00001.tif    0.395486666667    52.6995133333    -2.0
Calib_K00002.tif    0.395491666667    52.6995116667    -1.9

Actually this is gps coordinates recorded on the gps of the drone rather than the GCP on the ground. The GCP file for the opendronemap should also have pixel location. This is the reason I haven't use OpenDroneMap for our work.

Regards

Dr. Bo LI
Image analysis specialist & Bioinformatician | Genetics and Crop Improvement
NIAB EMR | New Road, East Malling, Kent, ME19 6BJ
01732 523734   bo.li@emr.ac.uk

dakotabenjamin, 2016-06-16:
Yes the coords in a photo's EXIF are the coords of the camera as well. In ODM, these coordinates are references only. I was thinking that we could assume that the drone coords are equal to the camera coords, then write those to the ODMPhoto object as if they were EXIF data.

dakotabenjamin, 2016-06-16:
How do you think we could improve ODM so you can use it for your work more easily?

tianyufang1958, 2016-06-16:
Hi Dakota,

We imaged a large wheat field, which included 400 images, but there are only 5 GCP. We have the list of all the gps coordinates as the txt file. So it would be helpful that opendronemap can also read our format as reference like Agisoft photoscan. Do you think it is possible to write our gps coordinates to exif and run opendronemap with no GCP file.

Regards

Dr. Bo LI
Image analysis specialist & Bioinformatician | Genetics and Crop Improvement
NIAB EMR | New Road, East Malling, Kent, ME19 6BJ
01732 523734   bo.li@emr.ac.uk

dakotabenjamin, 2016-06-16:
With python and a tool called [GExiv2](https://wiki.gnome.org/Projects/gexiv2), you could write the exif manually. In the future this will be a feature of ODM.

tianyufang1958, 2016-06-16:
Thanks Dakota, I will give it a go and let you know.

Regards

Dr. Bo LI
Image analysis specialist & Bioinformatician | Genetics and Crop Improvement
NIAB EMR | New Road, East Malling, Kent, ME19 6BJ
01732 523734   bo.li@emr.ac.uk

smathermather, 2018-03-23:
As this is a relatively old issue, I will close for now. Please feel free to reopen if necessary.

closed 2018-03-23 by smathermather