OpenDroneMap/ODM #310: Good or Bad dataset

PaoloCaf, 2016-05-18:
Hi,

I'm trying to use opendronemap with my dataset.
For some reason the final point cloud is empty and the reconstruction doesn't complete. 

######
I get the following error:

running "/home/ubuntu/OpenDroneMap/bin/odm_meshing" -inputFile /home/ubuntu/surv                                     ey/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /home                                     /ubuntu/survey/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -lo                                     gFile /home/ubuntu/survey/reconstruction-with-image-size-2400/odm_meshing/odm_me                                     shing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverD                                     ivide 9
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input clo                                     ud!
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[pcl::io::savePLYFile] Input point cloud has no data!
Error in OdmMeshing:
Error when saving mesh to file:
/home/ubuntu/survey/reconstruction-with-image-size-2400-results/odm_mesh-0000.pl

The pmvs didn't report any issue but the option-0000.py is empty.

##############

How can I troubleshoot this problem and find if there is a problem on my dataset?

thanks,

Paolo

dakotabenjamin, 2016-05-18:
How many photos are in your dataset? Can you post a sample photo? Can you post the stdout of the following call:
```
ls -l /home/ubuntu/survey/reconstruction-with-image-size-2400/opensfm/
```

PaoloCaf, 2016-05-18:
Give me one day.
I did  it in aws and terminated  the instance.

PaoloCaf, 2016-05-19:
Ok ... ready.
I have 51 images.
This is a sample image:
https://www.dropbox.com/s/e0t9dzyf5d6jf2x/IMG_0879_geotag.JPG?dl=0
Images have been already renamed with small characters.

`ubuntu@ip-172-31-31-46:/mnt/OpenDroneMap$ ls -l /mnt/survey/reconstruction-with-image-size-2400/opensfm/
total 125808
-rw-r--r-- 1 root root  7082298 May 19 12:34 bundle_r000.out
-rw-r--r-- 1 root root      336 May 19 12:19 camera_models.json
-rw-r--r-- 1 root root      108 May 19 12:19 config.yaml
drwxr-xr-x 2 root root     4096 May 19 12:19 exif
-rw-r--r-- 1 root root     1173 May 19 12:19 image_list.txt
-rw-r--r-- 1 root root     1019 May 19 12:34 list_r000.out
drwxr-xr-x 2 root root     4096 May 19 12:29 matches
-rw-r--r-- 1 root root      138 May 19 12:34 profile.log
-rw-r--r-- 1 root root 19333944 May 19 12:34 reconstruction.json
-rw-r--r-- 1 root root 90482682 May 19 12:34 reconstruction.meshed.json
-rw-r--r-- 1 root root       79 May 19 12:29 reference_lla.json
drwxr-xr-x 2 root root    12288 May 19 12:20 root_hahog
-rw-r--r-- 1 root root 11723644 May 19 12:29 tracks.csv
`

dakotabenjamin, 2016-05-31:
1. you may have too few images
2. your overlap may be insufficient, leading to a sparse point cloud that cannot be densified in PMVS.

dakotabenjamin, 2016-06-15:
Any luck @PaoloCaf?

PaoloCaf, 2016-06-15:
Hi,

Unfortunately I had no chance to test it again due to a crash and then bad
weather.
This weekend the forecast says that it will be sunny and I will most
probably fly again and let you know.

thanks a lot.
Paolo

rionlerm, 2016-07-08:
Hi @PaoloCaf and others,

I recently found sidelap below 60% and overlap below 80% does not produce results in ODM :( Also, here are some LAS file results (converted to Digital Surface Model) on top of the orthophoto created by OpenDroneMap (ODM). As you can see the dotted black line shows only two transect lines that were flown at 120 metersATL. This was not adequate for the proper creation of an elevation file in ODM.

[image]

dakotabenjamin, 2016-07-08:
Great work @rionlerm this is good information to know. When the side/overlap is too low, where does ODM fail?

smathermather, 2016-07-11:
Yes @rionlerm, it would be good to know if it fails at sparse point cloud creation or dense (or somewhere else).

rionlerm, 2016-08-04:
Hi @dakotabenjamin  and @smathermather ,

I apologise for never replying to your last questions.

I accidentally calculated transect width incorrectly with a flight and resulted in prob 40% overlap and 30% sidelap. ODM produced the following folders and contents **only**:

.
├── images
├── images_resize
├── opensfm
│   ├── exif
│   ├── matches
│   └── root_hahog
└── pmvs
    └── recon0
        ├── models
        ├── txt
        └── visualize

I think the script paused at `Expand patches...`

Please mention if you need to see any additional files in these directories.

dakotabenjamin, 2016-08-04:
Anything under 60% is going to be hard to reconstruct. You can try using the parameter `--pmvs-minImageNum 2` which means that PMVS will only require 2 matching images to reconstruct rather than 3. 

Otherwise, you may be able to play around with other cmvs-pmvs parameters to get it to run without crashing. [This is a resource I use](http://www.di.ens.fr/pmvs/documentation.html) often to try to get things working the way I want.

rionlerm, 2016-08-05:
Thanks Dakota!

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/good-or-bad-dataset/407

closed 2018-03-28 by dakotabenjamin