OpenDroneMap/ODM #311: odm_data_rv_nir data running failed at pmvs2

jangbi882, 2016-05-19:
I downloaded odm_data_rv_nir-7c86404650f1ad2cddd1b48995b950d776be48a9.zip.
And run it using opendronemap/opendronemap:dev Docker.
But it is stopped on running /code/SuperBuild/install/bin/pmvs2 /images/pmvs/recon0/ option-0000.

I ran it two times because first try stopped with no clear message.

I attach log file and result files list.
[odm_data_rv_nir_result_files.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/271705/odm_data_rv_nir_result_files.txt)
[odm_data_rv_nir_log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/271706/odm_data_rv_nir_log.txt)

dakotabenjamin, 2016-05-31:
It looks like you ran out of memory. This seems to be a recurring problem with larger datasets: #296 #290 

This is a software that should be able to handle large numbers of images, that's its job. What are your machine's specs?

jangbi882, 2016-05-31:
8GB.
I will retry after supply more ram to my VM.

Thank you.

Regards
BJ Jang.

<https://github.com/OpenDroneMap/OpenDroneMap/issues/296> #290

closed 2016-06-09 by jangbi882

smathermather, 2016-06-10:
Did you have luck, @jangbi882?

jangbi882, 2016-06-10:
Yes. this work was finished good with 24GB RAM VM.
Thank you so much.