OpenDroneMap/ODM #312: GCPs + EXIF?

rionlerm, 2016-05-26:
Not so much an issue but rather a question.

If I pass `--odm_georeferencing-useGcp` (.txt list of GCPs), would the images' GPS-tagged data be considered at all in orthophoto generation?

I am asking because, of a concern about a current dataset's GCPs being placed too far apart and not contained in all images.

smathermather, 2016-05-27:
I believe (without double checking) that GCPs are an override. We'll be moving to a more integrated model soon which takes into account a weighted set of inputs including any geo info + matching info which should serve to do a better job overall of georeferencing, but for now it's kinda simplistic.

smathermather, 2016-05-28:
What this OpenSfM branch: https://github.com/mapillary/OpenSfM/tree/gcp

pierotofy, 2020-02-03:
It used to be that we used only 3 GCPs, but OpenSfM now allows us to use all of them during computation. I'll close this for now. :pray:

closed 2020-02-03 by pierotofy