OpenDroneMap/ODM #314: How to edit ccd_defs.json from docker installation

kikislater, 2016-05-30:
Hi,
How to edit or add information in ccd_defs.json with docker installation ?

dakotabenjamin, 2016-05-31:
Good question-- normally I would say submit a PR here (the image rebuilds after every pushed commit) but as per #237 we are not accepting PRs for ccd_defs. If your camera is not in the definitions list simply use `--force-ccd`.

kikislater, 2016-06-02:
ok thank you for --force-ccd, I miss it. But normally, in a docker container we can run a bourne shell and modify a file with it but bash seems to not working from ODM container : 
`docker exec -i opendronemap /bin/bash -c 'cat > /inside-container-file' < external_file`
It just launch run.py

So I found how it works : 
ENTRYPOINT in dockerfile launch by default run.py
If I want to enter in the container with bash, I must have to specify it with : 
`docker run -t -i --entrypoint="/bin/bash" opendronemap`
I can watch the file in /code/ccd_defs.json with cat but cannot edit because nano is not installed. So I have to run the container with root => 
`docker run -t -i --user root --entrypoint="/bin/bash" opendronemap`
`apt-get install nano`
`nano /code/ccd_defs.json`
Unfortunately installing nano is not persistent ... Still learning docker under archlinux, I have to investigate !

dakotabenjamin, 2016-06-02:
For one, you should be appending to SuperBuild/src/opensfm/opensfm/data/sensor_data.json. ccd_defs is depreciated. 

Second, you can use sed to append to the second to last line of a file without using nano. 
e.g.
```
sed '$i    "Camera Model": 1.23,' SuperBuild/src/opensfm/opensfm/data/sensor_data.json
```

smathermather, 2016-12-20:
@kikislater -- is this issue closed?

kikislater, 2016-12-21:
Yes, now I have more knowledge with docker. It just basic usage like using docker commit.
You could close

closed 2016-12-21 by dakotabenjamin

winsento, 2017-02-01:
Small changes @dakotabenjamin command:

```
sed '2i\    "Camera Model": 1.23,' SuperBuild/src/opensfm/opensfm/data/sensor_data.json
```