OpenDroneMap/ODM #319: Running the new install and getting an error

LeithHawkins, 2016-06-19:
Howdy,

just cloned the latest version and I am having trouble getting it to run ?

I am trying to run the run.sh file - and this is the error I get ?

I m running an EC2 instance of ubuntu 14.04

```
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap/run.py", line 8, in <module>
    import ecto
ImportError: No module named ecto
```

after we have cloned do we still need to run the install process - screen shot of whats happening below ?  

[image]

thanks

Leith

Fi156, 2016-06-19:
Hi Leith,

the new version is quite different to the old one. Have a look at this post here:
https://github.com/OpenDroneMap/OpenDroneMap/releases/tag/v0.1

There are also changes in the folder structure etc.:
https://github.com/OpenDroneMap/OpenDroneMap#running-opendronemap

LeithHawkins, 2016-06-19:
Thanks so much

closed 2016-06-19 by LeithHawkins