OpenDroneMap/ODM #322: Issue With Colour Blending

arosendorff, 2016-06-24:
Hi,

I've recently installed OpenDroneMap and am trying to use it to generate orthophotos. However, the final output image appears to be stitched together properly, but the colour is very patchy, changing in sections. Is there a way to disable this if it is intended? I cannot seem to find a way to enable a mosaic mode for generating the orthophotos, which is what I assume I need to prevent the colour blending.

Here's an example of th issue I'm having, the colour of the original images is all perfectly uniform.
[image]

Thanks!

dakotabenjamin, 2016-06-24:
Yes this is currently how the orthophoto is generated. Don't lose hope, though. We've got updates in the pipeline that improve texturing, implementing better blending. It's not ready for production yet, but the `mvs-texturing-addition` branch has what you need. See the image below for differences:

[image]

arosendorff, 2016-06-24:
Okay thank you, that does look better, I'll give it a try!

closed 2016-06-24 by arosendorff