OpenDroneMap/ODM #323: Error while installing

femape, 2016-06-25:
Hello,
I have 2 questions, I hope someone can help me:
1) Did someone of you ever install ODM in Arch Linux?
2) When installing in Ubuntu 16 (fresh virtual machine), the process stops after one warning and one error. I get the following message after executing "cmake ..", is this already known?:

**~/Downloads/OpenDroneMap/build$** cmake ..
CMake Warning at modules/odm_georef/CMakeLists.txt:16 (find_package):
Could not find a package configuration file provided by "PCL" (requested
version 1.7) with any of the following names:

PCLConfig.cmake
pcl-config.cmake

Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
to a directory containing one of the above files. If "PCL" provides a
separate development package or SDK, be sure it has been installed.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
Could not find a package configuration file provided by "OpenCV" with any
of the following names:

OpenCVConfig.cmake
opencv-config.cmake

Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
"OpenCV_DIR" to a directory containing one of the above files. If "OpenCV"
provides a separate development package or SDK, be sure it has been
installed.

-- Configuring incomplete, errors occurred!
See also "/home/winner/Downloads/OpenDroneMap/build/CMakeFiles/CMakeOutput.log".

dakotabenjamin, 2016-06-27:
I have not tried Arch Linux. 

I built a fresh virtual machine with 16 and could not reproduce your problem. Did you run into any errors during `bash configure.sh`? 

Check to see that OpenCV is in SuperBuild/build/opencv

femape, 2016-06-27:
Hi @dakotabenjamin, thanks for your help. 
When I first run the config script, I think there was no error but now I execute:
export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages:`pwd`/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib
bash configure.sh

Then I get the attached log, which says that there is an error regarding the PCL build.

[config_log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/335873/config_log.txt)

dakotabenjamin, 2016-06-28:
what is the output of `free -m`? It looks like you're having the same issue as https://github.com/OpenDroneMap/OpenDroneMap/issues/324

femape, 2016-06-28:
Thanks @dakotabenjamin, you're totally right, the problem was the memory. I increased to 4GB as you recommended in the referenced post and I was able to run the configuration + compilation.
Now I'm testing it with the seneca images ;)

closed 2016-06-28 by femape

yohanboniface, 2017-03-29:
@fedeman did you finally manage to install ODM in Arch or did you use an Ubuntu VM?

fedeman, 2017-03-29:
Hi @yohanboniface, 
I used a virtual machine with Ubuntu on it (don't forget to use enough RAM size), since ODM has many dependencies and I didn't want to invest much time while installing.

dakotabenjamin, 2017-03-29:
Sorry, I wish we had enough resources to devote to a native multi-platform solution. I would suggest using our docker deployment: https://github.com/OpenDroneMap/OpenDroneMap#build-and-run-using-docker

kikislater, 2017-03-30:
+1 for docker, it's more powerfull than a VM.
I use it on archlinux