OpenDroneMap/ODM #324: Error in configure.sh

smith981, 2016-06-27:
~~~
...
[ 29%] Building CXX object io/CMakeFiles/pcl_io.dir/src/vtk_lib_io.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
io/CMakeFiles/pcl_io.dir/build.make:446: recipe for target 'io/CMakeFiles/pcl_io.dir/src/vtk_lib_io.cpp.o' failed
make[5]: *** [io/CMakeFiles/pcl_io.dir/src/vtk_lib_io.cpp.o] Error 4
CMakeFiles/Makefile2:248: recipe for target 'io/CMakeFiles/pcl_io.dir/all' failed
make[4]: *** [io/CMakeFiles/pcl_io.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/pcl.dir/build.make:113: recipe for target 'pcl/stamp/pcl-build' failed
make[2]: *** [pcl/stamp/pcl-build] Error 2
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/pcl.dir/all' failed
make[1]: *** [CMakeFiles/pcl.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Script finished
~~~

dakotabenjamin, 2016-06-27:
This looks like a compiler error, not an ODM error. Have you tried reinstalling the compiler?

smith981, 2016-06-27:
Sorry, I'm pretty green about compiling my own code. How would I do that?

dakotabenjamin, 2016-06-27:
I would just do 
```
sudo apt-get remove gcc-5
sudo apt-get install gcc-5
```

smith981, 2016-06-27:
Thanks for replying so quickly, but unfortunately I still get the same result after following your instructions.

dakotabenjamin, 2016-06-27:
What's the output of `free -m`?

smith981, 2016-06-27:
~/OpenDroneMap$ free -m
                    total        used        free      shared  buff/cache   available
Mem:            975         534         217           0         223         384
Swap:          1021         567         454

smith981, 2016-06-27:
Hey, look at that! Increased memory and HD space for my VM and now I'm past that point. Thanks guys!

closed 2016-06-27 by smith981

dakotabenjamin, 2016-06-27:
Great! you'll want more memory as well (at least 4) to run ODM without crashing

smith981, 2016-06-27:
You rock, @dakotabenjamin. Much thanks!