OpenDroneMap/ODM #325: Texturing  and orthophoto options

Fi156, 2016-06-27:
Hello,

there are two options regarding the texturing step:

>  --odm_texturing-textureResolution <positive integer>
>                       The resolution of the output textures. Must be greater
>                       than textureWithSize. Default: 4096
> --odm_texturing-textureWithSize <positive integer>
>                       The resolution to rescale the images performing the
>                       texturing. Default: 3600

And one regarding the orthophoto creation:

> --odm_orthophoto-resolution <float > 0.0>
>                       Orthophoto ground resolution in pixels/meter. Default:
>                       20.0

To get the best quality possible, I will of course set the orthophoto parameter to a cm value. For Example: 20 Pixels/meter = 0.05 Meter/Pixel (GSD)

For the other two parameters: 
Are they affecting the orthophoto creation at all?
Can I take the original resolution from the input images? 
Does these values have to stay under the the value of the "resize to" parameter?

Thanks :)

dakotabenjamin, 2016-06-29:
AFAIK, the resolution of the textured model directly affects that of the orthophoto. 

I do believe you have to specify manually the original photo resolution for those parameters, and they do not necessarily have to be under the --resize-to number.

pulquero, 2016-12-13:
Is it possible to have some documentation about these parameters? I'm not entirely sure I understand what effects they all have. For instance, I believe that you need to set --resize-to small enough to smooth out and remove background/misc features but large enough to preserve the main features. (Conversely, --min-num-features will need to be larger for larger resized images because you are going to hit more misc features during feature extraction). And that the resized images are only used for feature extraction and don't effect the quality of any texturing.

Now I assume that with --odm_texturing-textureResolution and --odm_texturing-textureWithSize you can somehow control the number of textures produce? The more the better?

dakotabenjamin, 2016-12-14:
@pulquero these parameters don't exist anymore because we've replaced the odm_texturing module. The new external module (mvs_texturing) uses the 3D scene generated by PMVS (or opensfm, is you use the depth mapping approach) with the resized images.

pulquero, 2016-12-14:
Oh, you're right, just had a look in config.py. I always used the wiki for reference, but that is now quite out-of-date.

dakotabenjamin, 2016-12-14:
Yup. I just updated the wiki. You can always run `python run.py --help` for the parameters. 

I don't think there is any reason to use the old odm_texturing module. It's on my list of things to purge from the codebase.

closed 2018-03-28 by dakotabenjamin