OpenDroneMap/ODM #327: configure.sh issue

ju1998ly, 2016-06-28:
Hello I am stuck on step 11. After "bash configure.sh" i get these errors at the end. 
[image]

dakotabenjamin, 2016-06-29:
It's hard to say what is failing from that screenshot because it is a multi-threaded compilation.

However, this is a recurring issue: https://github.com/OpenDroneMap/odm_vagrant/issues/15

ju1998ly, 2016-06-29:
Yes, it seems we have the same issue.

ju1998ly, 2016-06-29:
This is just a guess but some files may have not install correctly because some variable were undeclared. Such as meanStore

ju1998ly, 2016-06-30:
Question: Do we make the python path before or after the bash configure.sh? And are you Abel to produce the same results? I'm using Ubuntu 14.04.

dakotabenjamin, 2016-06-30:
You must set the environment variables (PYTHONPATH and LD_LIBRARY_PATH) before running the python script run.py.

dakotabenjamin, 2016-06-30:
I've been unable to replicate this issue on a fresh vagrant machine. (SInce you are referring to step 11, I am assuming you are following the directions for setting up a vagrant machine)

 I need the whole log of your install in order to properly understand the problem you are facing.

ju1998ly, 2016-06-30:
Are you talking about the steps I used to install or the logs of the vagrant machine? If it is the logs I am not sure how to get them. Can you tell me how? And, Thank you.

ju1998ly, 2016-06-30:
[image]

dakotabenjamin, 2016-06-30:
2 ways: 

1. If you're in git shell or git bash, you can simply highlight and copy the text into a document on your host machine. 

2. you can run the configure.sh script to write to a file:
```
bash configure.sh 2>&1 | tee config.log
```
And then post the log to pastebin or something like that.

ju1998ly, 2016-06-30:
Okay

ju1998ly, 2016-06-30:
I created the log file but how do i post it or access it from windows?

dakotabenjamin, 2016-06-30:
You can `cp config.log /vagrant_data/` and then you should be able to access it from that vodm_data folder you made before.

dakotabenjamin, 2016-06-30:
Then you can upload it to http://pastebin.com/ or https://gist.github.com/

ju1998ly, 2016-06-30:
The file did not show up there. Are there other commands?

dakotabenjamin, 2016-06-30:
what's the output of `ls -al /vagrant_data/` ?

ju1998ly, 2016-06-30:
It says that it cannot be access and that there is No such file or directory.

dakotabenjamin, 2016-06-30:
Hmm... did you follow the steps here https://github.com/OpenDroneMap/odm_vagrant ?

The Vagrantfile should connect `/vagrant_data/` to `vodm_data` without any issue.

ju1998ly, 2016-06-30:
I will retry the steps again.

heenalee, 2016-12-05:
I am stuck on Step #11 as well. I've tried a few times to install OpenDroneMap. My system is a multi-threaded compilati
[image]
on. What can I do? Please help!

dakotabenjamin, 2016-12-06:
@heenalee I don't see any error in your screen grab. Can you find the error in the configure.sh output and post that?

heenalee, 2016-12-06:
How can I do that? I am a newbie...

heenalee, 2016-12-06:
I restarted to run "bash configure.sh" and got stuck again. Please see attached. Is this what you are looking for?
[Stuck on Bash Configure.sh.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/634083/Stuck.on.Bash.Configure.sh.txt)

pierotofy, 2016-12-06:
@heenalee delete the build/ and SuperBuild/build directories. Then restart the configure.sh script.

To capture all output, run:

`bash configure.sh 2>&1 | tee -a log.txt`

Attach the log.txt file if you can't build.

heenalee, 2016-12-06:
Thank you pierotofy to add the comment!

I've found the reason it got stuck. The VM ran out of the memory. I've changed the memory setting in VirtualBox Manager, but it overwrites when I run Vagrant Up. So I've changed the memory setting in Vagrantfile directly. 
Now, it ran through Configure.sh but got errors. Please take a look attached file. 
[Configure.sh Error.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/635270/Configure.sh.Error.txt)

pierotofy, 2016-12-07:
Well, it looks like you might still have some memory issues...

```
virtual memory exhausted: Cannot allocate memory
/usr/bin/cmake: error while loading shared libraries: libc.so.6: cannot map zero
-fill pages: Cannot allocate memory
```

How much RAM did you allocated to the VM?

heenalee, 2016-12-07:
4096 out of 8192. I watched my computer memory went up to 70% and cpu usage went up to 100%.  
Do I need to modify the cpu as well? vb.cpus = 9 in Vagrantfile.

pierotofy, 2016-12-07:
Try allocating more memory, or enable a swap file before starting the compilation process.

closed 2018-03-28 by dakotabenjamin