OpenDroneMap/ODM #328: python: can't open file 'run.py': [Errno 2] No such file or directory

Bavar2142, 2016-06-29:
:~$ python run.py --project-path /home/Pictures
python: can't open file 'run.py': [Errno 2] No such file or directory

Anyone got ideas as to how i'd get around the above issue?

Bavar2142, 2016-06-29:
After restarting it is now saying permission denied

/home/gis/Pictures# ../OpenDroneMap/run.py
bash: ../OpenDroneMap/run.py: Permission denied

Fi156, 2016-06-29:
try:
home/gis/Pictures# `python ../OpenDroneMap/run.py`

Bavar2142, 2016-06-29:
~/Pictures$ python ../OpenDroneMap/run.py
Traceback (most recent call last):
  File "../OpenDroneMap/run.py", line 8, in <module>
    import ecto
  File "/home/gis/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

is what i get. Does that help?

Bavar2142, 2016-06-29:
I have the images themselves within a folder called images in the Pictures folder.

Bavar2142, 2016-06-29:
This is what I get when I try to run it in the images folder:

/Pictures/images$ python ../OpenDroneMap/run.sh
python: can't open file '../OpenDroneMap/run.sh': [Errno 2] No such file or directory

Bavar2142, 2016-06-29:
I am currently running a clean install of Ubuntu 16.04 if that helps.

dakotabenjamin, 2016-06-29:
If I understand correctly, your ODM install is here: `/home/gis/OpenDroneMap`

If that is the case, then it would be easier to `cd /home/gis/OpenDroneMap` then 
```
./run.sh --project-path /home/gis/Pictures`
```

Bavar2142, 2016-06-29:
Bingo. Thanks for all the help and patience.

closed 2016-07-01 by smathermather

dakotabenjamin, 2016-07-01:
@Bavar2142 it looks like the problem was that you have to call run.sh from within the ODM directory, that's all.