OpenDroneMap/ODM #329: Bad result when changing default resize-to parameter

rmallermartins, 2016-06-29:
I'm having an issue with v0.1 when using bigger resize-to parameter then the default value. 

Result with the default value:
[image]

Result with resize-to equals 3000:
[image]

Both are taking +- 30min to process

dakotabenjamin, 2016-06-30:
OpenSfM tries to create a single reconstruction, but sometimes needs to make multiple, and pmvs will only take the largest reconstruction. See this issue for more technical explanation: https://github.com/OpenDroneMap/OpenDroneMap/issues/186

smathermather, 2016-07-01:
I don't know if this is the issue, or it's a matter of tweaking the OpenSfM/PMVS parameters to increase the match number. Any thoughts @paulinus?

dakotabenjamin, 2016-07-20:
Try increasing `--min-num-features` to increase image matches.

closed 2018-03-28 by dakotabenjamin