OpenDroneMap/ODM #330: Error 32512 During ODM Mesh file writing

Bavar2142, 2016-06-30:
[ERROR]   quitting cause: 
	/home/gis/OpenDroneMap/build/bin/odm_meshing -inputFile /home/gis/Pictures/pmvs/recon0/models/option-0000.ply -outputFile /home/gis/Pictures/odm_meshing/odm_mesh.ply -logFile /home/gis/Pictures/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
returned with code 32512.

Above error occurred during mesh generation of Ubuntu 16.04. I5 quad core, 4 gigs of ram and ssd.

dakotabenjamin, 2016-06-30:
What's in this: /home/gis/Pictures/odm_meshing/odm_meshing_log.txt

Bavar2142, 2016-07-03:
Ill run it again and get back to you

Bavar2142, 2016-07-03:
Its also saying could not find ccdwidth file.

closed 2016-07-05 by Bavar2142

jpmaniac, 2016-11-12:
I'm having the exact same issue on the dev branch... have you found out what caused it or a work around?