OpenDroneMap/ODM #332: Ubuntu 14.04 32GB, 6 core XEON, fails with 'returned with code 33792.'

robham101, 2016-07-02:
Hi,

Fairly new to OpenDroneMap so I am sure this is most likely user error but I am getting an error when I try to generate a single 2d image stitched together from my JPG's.

I have around 1000 JPG files, each has been taken with the camera pointing directly downwards and has its GPS co-ordinates in its EXIF data. I have created a folder called HAV, put the JPG files into a subfolder called images. Then from the OpenDroneMap folder one level above I have ran the command:

**./run.sh --project-path ../HAV --force-ccd 17.3**

I am running Ubuntu 14.04 as a virtual machine on MAC OSX 10.10.
It is pretty fast - has 6 core 3.9GHz XEON Processor, 64GB Memory, and a 4TB RAID Array consisting of 8 x 512GB SSD Drives in a RAID Stripe array so it resizes the images and processes the data reasonably quickly, listing lots of Full matching lines, and Robust matching time etc.

It creates an opensfm folder, in the matches subfolder there seems to be a file for every one of my photos, there are 2814 files in the root_hahog subfolder which is 4.7GB. The tracks.csv file in the opensfm folder is 150Mb.

It creates a PMVS folder - but it is empty.
### 
**The final output on the screen is:** 

DJI_0033.JPG - DJI_0981.JPG has 100 candidate matches
Robust matching time : 0.000235795974731s
Full matching 97 / 100, time: 0.953299045563s
DJI_0033.JPG - DJI_0980.JPG has 97 candidate matches
Robust matching time : 0.000231981277466s
Full matching 93 / 97, time: 0.874423027039s
2016-07-02 21:53:40,945 reading features
2016-07-02 21:53:46,929 Merging features onto tracks
2016-07-02 21:54:10,922 Good tracks: 773584
images 934
nonfisheye images 934
Initial reconstruction with DJI_0533.JPG and DJI_0534.JPG
Number of common tracks 1825
/home/parallels/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all: line 11: 25488 Illegal instruction     (core dumped) $DIR/reconstruct $1
[ERROR]   quitting cause: 
	PYTHONPATH=/home/parallels/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/parallels/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/parallels/HAV/opensfm
returned with code 33792.

An error occurred. Check stdout above or the logs.

I have downloaded the 'benchmark' sample data and this processes OK, so it must be something in my data or something I am doing wrong.

Any assistance getting it to work would be greatly appreciated.

Kind regards,

Rob

Fi156, 2016-07-03:
Looks like an OpenSfM error. @paulinus  do you have an idea?

I would try to make a subset from the images:
1. You can test something quickly without waiting a long time. 
2. There are memory related issues further down the process and I dont know if 64GB ram will be enough to process 1000 images.

Btw: Good camera choice ;)

robham101, 2016-07-03:
Thanks everso much for the response Fi156. I did try knocking the image count down to around 300, but I will try extracting just a very small subset and see what happens with those.
Yes very impressed with the camera, although finding it  a bit hit and miss when it comes to selecting aperture in A or M mode as sometimes you have to restart the Inspire several times before the option to change aperture is available, most of the time it is greyed out - thats an issue for another day though!
I will try a smaller selection of data in OpenDroneMap and let you know what happens!
Thanks again for the help.
Rob

Fi156, 2016-07-03:
64GB RAM and 300 Photos is fine. You dont need to choose a smaller subset...

Another shot in the dark: Try to reduce the image size with `--resize-to 800` and/or `--min-num-features 1200`.

How much overlap do you have between the photos?

robham101, 2016-07-03:
I will give the resize a try. I have also just realised I do not have 64GB RAM. The host machines does, but the virtual linux box only has 32Gb allocated to it at the moment.

Bavar2142, 2016-07-03:
I had issues with one data  set i ran through it, though that could be due to the specs of the machine. Sub-question if thats allowed. What specs would you recommend to run the software for up to 1000 image datasets?

rionlerm, 2016-07-05:
Lowering the resolution of the orthophoto (stage where memory quitted cause: orthophoto creation) from the default 20 pixels/meter to `--odm_orthophoto-resolution 5.0` (20cm on ground) resolved my issue. I only have 4GB RAM and between 100 and 200 photos that were resized by default.

dakotabenjamin, 2016-07-05:
I don't really know what to recommend @Bavar2142 but the limiting factor in most cases right now is RAM (so long as you're willing to wait). You may run into issues during PMVS and orthophoto generation. I run big stuff with 42 GiB/42 Gib swap with no problem. I *think* you should be able to do it with as little as 16. Like @rionlerm said, though, you can work with the parameters to make it work. 

---
@robham101 let's see if you can reconstruct with a subset of photos

Fi156, 2016-07-05:
@rionlerm as far as I undertsand the error was thrown alredy in OpenSFM, so neiterh PMVS nor the Orthophoto creation is involved... Thats the reason, why I suggested changes in the parameters for the matching step...

Bavar2142, 2016-07-05:
At the moment we're just using a word processor box to do the processing until we can get a new rig. Thanks for the heads up on the ram amount thatll give me an idea of how much id need. How would I go about adjusting the parameters in the matching step? @dakotabenjamin @Fi156

dakotabenjamin, 2016-07-06:
Like @Fi156 said above, reduce `--min-num-features` and/or `--resize-to`. The resize difference won't affect the texturing and orthophoto image resolutions, only the point cloud generation parts of the toolchain.

dakotabenjamin, 2016-07-22:
@robham101 I know the thread got a little derailed, did you solve your problem?

robham101, 2016-07-23:
Thanks for the help everyone, and for checking on progress @dakotabenjamin 
In the end I used a program called Autopano Giga Pro to do the stitching as I ran out of time, but at some point I will take a look at it again and give it another go using OpenDroneMap.

closed 2016-07-23 by robham101

alexhagiopol, 2016-08-26:
I am actually having this exact same issue running ODM in a Docker image on a Xeon server. Anyone learn anything about this since July?

This is the error that happens during the OpenSFM step:

```
OpenDroneMap/SuperBuild/src/opensfm/bin/run_all: line 11: 25488 Illegal instruction (core dumped) $DIR/reconstruct $1
```