OpenDroneMap/ODM #333: Recommend Hardware?

Bavar2142, 2016-07-07:
Hello all, 
apologies if this is a daft question and feel free to move to appropriate area if this isnt the place.  Does anyone know if the ghz of the cpu or the number of cores is more important for ODM? We're looking at building a processing rig on this end  and are thinking of 64 gig ram and a samsung 850 pro ssd.

alexhagiopol, 2016-07-07:
Hi @Bavar2142 I've been working extensively with ODM and other 3D reconstruction libraries. Here's a list of parts in my development rig. Let me know if you have questions or if you're thinking of building the same one. The only changes I recommend are (1) maybe getting a cheaper GPU and (2) getting a single set of 64GB of RAM as opposed to two sets of 32GB.

http://pcpartpicker.com/list/sYcPsJ

Good luck! 

Alex

dakotabenjamin, 2016-07-07:
I don't know if I can recommend GHz over cores, but on my work computer (not the server we do production stuff on) I have an Intel Xeon CPU E5-1650 (3.5Gz) and 32 GB memory. We just installed an Nvidia 1080 (stock) but right now ODM does not support OpenCL or any GPU processing.

alexhagiopol, 2016-07-07:
Just realized I didm't answer the question. I'd say GHz and cores are about equally important. That's why I went the i7 route and overclocked to 4.3GHz on 6 cores. Highly recommend the 850 Pro SSDs as well.

GeospatialDaryl, 2016-07-07:
I'm a Samsung SSD fanboy as well - but for almost all applications, the
EVOs are fine.  The Pros are significantly more expensive.  Unless you are
doing server database transactions, it's unlikely you'd run it hard enough
to see the long-term performance difference between the Pro and EVO.   One
technology I'd look for in a motherboard is support for NVMe boots - both
in the PCIe form factor and M.2 (NVMe, not SATA) on the motherboard.
Wicked fast.

While intel continues to dominate for instructions per clock, there is a
substantial value (still, especially for home lab crunch boxes) in the
aging AMD FX line.  If you have the budget, those Xeons are sure nice - but
take a close look at the FX 8320e (8 cores for ~$130).  Mine overclocked to
4GHz without any fiddling - on a Corsair Hyper 212 EVO.  I'm looking to get
it up to 4.3 or so when I have an afternoon for that.  For my applications,
more cores is better than more IPC.  And the MBs and CPUs are cheap these
days.    If you have a few months, I'd wait for AMD Zen.

Just my 2 cents-

Daryl

Bavar2142, 2016-07-08:
Hello all,
thanks for all the replies. In terms of the cpu I have a 6600K on my home rig that ive gotten up to .43 ghz. Would that approach be worth a look? Hows the 1080 treating you? @dakotabenjamin 

In terms of your build @alexhagiopol at the moment the changed id make if I was building today would be H110 or 115 for AIO cooler, a 700 series gpu (just to save money with the processing being cpu dependant) WHat made you recommend a single stick of ram?

Bavar2142, 2016-07-08:
In terms of time table itll be a while as we are waiting on hiring a new it person so zen may be an option.

Bavar2142, 2016-07-08:
@dakotabenjamin Do you know if there are any plans to share some of the processing load with the GPU eventually?

Fi156, 2016-07-08:
Pull request will be very welcome, but as far as I know, nothing is planned yet...

Bavar2142, 2016-07-08:
@Fi156 If I had the programming skill id love to do that. Sadly im just getting back into it after a long absence.

alexhagiopol, 2016-07-08:
@Bavar2142 If you look at the amazon link I posted for the RAM, you'll see it's 8 stricks of 8GB each. It's important to get them all in the same set because they are tested and validated together at the factory.

I don't tend to recommend the all-in-one liquid coolers because they are louder and have more points of failure than a good air cooler.

dakotabenjamin, 2016-07-08:
GPU processing is certainly in the future but right now we don't have any timeline. When we build and integrate semi-global matching [that would be a prime candidate for GPU processing](http://www.nvidia.com/content/events/sc11/pdf/ke-zhu.pdf)

Bavar2142, 2016-07-10:
Thnaks for the heads up @dakotabenjamin Thanks for the explanation @alexhagiopol I had originally thought you meant a single stick of ram. Ill paste a link to the parts list that we're looking at once I find it.

edgarriba, 2016-07-28:
some months ago I started a GPU based approach for features detection and matching in OpenSfM. Unfortunately, my schedule has changed and I cannot spend time on this right now. Probably yes in the future. However, if someone wants to follow with that PR will be great!

https://github.com/mapillary/OpenSfM/pull/49

closed 2016-09-28 by Bavar2142