OpenDroneMap/ODM #334: ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

tdecae, 2016-07-08:
Hi, 
I would really appreciate some help please. I'm trying to run ODM on a few pictures that I want to stitch together. My 'images' folder is in /home/ubuntu/OpenDroneMap/ so I run the following command from the /home/ubuntu/OpenDroneMap/ folder  
sudo python run.py /home/ubuntu/OpenDroneMap

but i get the following error:

sudo: unable to resolve host drone-image-processing
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/ubuntu/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory
ubuntu@drone-image-processing:~/OpenDroneMap$

if anyone can help me it would be much appreciated.

Thank you

dakotabenjamin, 2016-07-08:
Ach sorry this was a problem I meant to fix earlier. It's actually a really simple solution, I just haven't had any time to dedicate to it. You could easily fix it yourself:

On line 30 of `opendm/types.py` it now shows this:
```
        self.make_model = None
```

Replace `None` with `''`: (two single apostrophes)
```
        self.make_model = ''
```

tdecae, 2016-07-09:
Hi,
Thank you for your help and making this wonderful project also. I will try this fix early next week when back in office.
Best regards
Thierry

Sent from BlueMail<http://www.bluemail.me/r>

fahmiroi, 2016-07-29:
Hi,

Replacing None with ' ' didn't solve the problem for me. :(

Still have the same error message..

dakotabenjamin, 2016-07-29:
In #348 I have fixed this problem, I believe. Can you test for me?

```
git checkout -b dakotabenjamin-odmphoto_fix dev
git pull https://github.com/dakotabenjamin/OpenDroneMap.git odmphoto_fix
```

If you using the `dev` branch already, you will not have to rerun the installation procedures. If you do not have the `dev` branch installed, you may have to checkout dev first, then do the above, then run through the installation procedures.

fahmiroi, 2016-07-29:
It worked ! The problem is solved and the processing of my data set ist running :)

Many Thanks !!

dakotabenjamin, 2016-07-29:
Great news!

@tdecae would love if you tested as well.

troymyname, 2017-02-10:
@dakotabenjamin I am still getting the error. The _self.make_model =_ is already set to two single apostrophes. I would really appreciate it if you can help please.

dakotabenjamin, 2017-02-10:
I'm sorry to hear that @troymyname. Are you using the latest release? [v0.2](https://github.com/OpenDroneMap/OpenDroneMap/releases/tag/v0.2)

troymyname, 2017-02-10:
Hi, yes. I am using Version 0.2. Thanks for eating back with me.

dakotabenjamin, 2017-02-10:
Can you open a new issue with the error message you are getting, along with other relevant info, like project setup, etc?

troymyname, 2017-02-10:
@dakotabenjamin I have opened up issue #480.

closed 2018-03-28 by dakotabenjamin