OpenDroneMap/ODM #337: ImportError: No module named ecto_main

jamiephillips0000, 2016-07-11:
Hi 
when running the test I get the following error.
(OpenDroneMap) jamie@PUTER:~/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap$ python run.py --project-path ../odm_data_seneca/
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/jamie/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap/SuperBuild/install/lib/python2.7/site-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: No module named ecto_main

the offending line in "/home/jamie/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap/SuperBuild/install/lib/python2.7/site-packages/ecto/__init__.py" is
from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup

There seems not to be a file called ecto_main?
Any help is appreciated

Jamie

dakotabenjamin, 2016-07-11:
How did you set your environment variables?

jamiephillips0000, 2016-07-11:
.bashrc
export ODMPATH=/home/jamie/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap
export PYTHONPATH=$PYTHONPATH:$ODMPATH/SuperBuild/install/lib/python2.7/dist-packages:$ODMPATH/SuperBuild/src/opensfm:$ODMPATH/SuperBuild/install/lib/python2.7/site-packages/
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$ODMPATH/SuperBuild/install/lib

dakotabenjamin, 2016-07-11:
Ok it might have been an install error, if it doesn't throw a "can't find libecto.0.6" error. Try running `bash ./configure.sh` again and pay attention to `make` errors. You may want to edit https://github.com/OpenDroneMap/OpenDroneMap/blob/master/configure.sh#L123 to remove the `-j${NUM_CORES}` flag.

jamiephillips0000, 2016-07-11:
running bach ./configure now
Just looked at the ecto github and the code is exactly the same. Any chance the error comes from them?

dakotabenjamin, 2016-07-11:
It's a possibility but we would have to figure out why it's going wrong on your machine specifically first to determine that.

jamiephillips0000, 2016-07-11:
I added the output from the ./configure
Nothing seemd to have changed much
[out.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/357110/out.txt)

jamiephillips0000, 2016-07-11:
as far as I can see from the ecto project https://github.com/plasmodic/ecto
the file python/ecto/__init__.py makes a call to
from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup

and I cant find ecto_main anywhere
Also there is a test.py in the same folder and it fails at the same line
I went looking to see if there was a previous branch I could use - but with names like "amoeba" I am not encouraged ;-)

jamiephillips0000, 2016-07-11:
if this is really not an error from ODM just let me know and I will open it on the ecto project?

dakotabenjamin, 2016-07-11:
The configure output looks fine. Try one more thing for me: use `./run.sh` instead of `python run.py` (everything else the same) and see what happens. 

I don't know if the ecto people will be able to help because they don't know our code base, but I also don't think it would hurt to drop a line out there. Maybe @edgarriba could also provide some insight into this problem?

jamiephillips0000, 2016-07-11:
`./run.sh` produces the same error
```
(OpenDroneMap) jamie@PUTER:~/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap$ ./run.sh --project-path ../odm_data_seneca/
Traceback (most recent call last):
  File "/home/jamie/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap/run.py", line 8, in <module>
    import ecto
  File "/home/jamie/git/CODE/PYTHON/OpenDroneMap1/OpenDroneMap/SuperBuild/install/lib/python2.7/site-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: No module named ecto_main
```
As I'm looking into this, it has IMO, nothing to do with you. The seems to be an ecto problem. I will post it there and see what happens

dakotabenjamin, 2016-07-11:
OK. Let us know here what you find here.

edgarriba, 2016-07-11:
@vrabaud @ethanrublee is this any known ecto issue?

jamiephillips0000, 2016-07-11:
created the [issue in ecto](https://github.com/plasmodic/ecto/issues/286)

jamiephillips0000, 2016-07-12:
Hi Guys
I took this over from someone else and I now have new information - don't know if it will help.
1) Python is being used through an Anaconda install for the user. So I don't think **sudo pip install XXX** will do much if the files are not on the PYTHONPATH
2) apparently `--layout=deb` was not recognized so it was manually removed from the source files << the `bash ./configiure.sh` fails if this is taken out so I'm not sure this is a mistake.
Apologies for not disclosing this earlier - I just found out

dakotabenjamin, 2016-07-12:
1. I've only used Anaconda twice. I don't know how it would affect the build. Do you mean that Anaconda flavored python (I'm not sure how to call it) would interfere with the install process for python modules? Are you using anaconda or is the user?
2. Which source files? I can't find that anywhere.

jamiephillips0000, 2016-07-12:
Ok fixed it
Anaconda is the issue - it cant work with it
1) Anaconda Python install would affect ODM at run-time. You would have inconsistent dependencies that might lead to trouble. Anaconda installs python for a specific user and allows more control because you don't change versions with a system package manager update. 
Anaconda puts the following into the `.bashrc`
```
# added by Anaconda2 4.1.0 installer
export PATH="/home/jamie/anaconda2/bin:$PATH"
```
2) Setuptools in Anaconda does not recognize --install-layout=deb
ubuntu@odm:~/git/OpenDroneMap$ grep -R "layout=deb" SuperBuild/src/ 
SuperBuild/src/catkin/cmake/python.cmake:  set(SETUPTOOLS_ARG_EXTRA "--install-layout=deb")

I rolled a new VM with a new checkout and it works like a charm
Apologies for the distraction. Again, I was handed this to fix with little information

closed 2016-07-12 by jamiephillips0000

dakotabenjamin, 2016-07-12:
Not a distraction, now we know how to troubleshoot this problem in the future.

muzo583, 2016-08-12:
Hi jamiephilips00000
I have same problem and I have installed Anaconda on my desktop. Can you please clarify your solution with details for ecto problem? what is the meaning VM?
did you uninstalled anaconda?
thanks.

jamiephillips0000, 2016-08-12:
To disable anaconda you remove the following from your ~/.bashrc
``# added by Anaconda2 4.1.0 installer
export PATH="/home/jamie/anaconda2/bin:$PATH"``

I found that, for whatever reason, I just could not get this to work on my
existing machine. Maybe I had some python stuff from before - I just don't
know and didnt take the time to find out. What worked was that I created a
vmware player VM with a vanilla install of Ubuntu 14.04 and it worked like
a charm.
Hope this helps

rajsaurav1234, 2018-04-19:
ubuntu@ubuntu-ESPRIMO-P420:~/IIRS/OpenDroneMap$ python run.py -i /home/ubuntu/ROV image/1 yhyh
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
ImportError: No module named ecto