OpenDroneMap/ODM #338: Mesh construction

v4ven27, 2016-07-11:
Greetings ODM users, 

I've been looking into the ODM source code, particularly [OdmMeshing.cpp](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_meshing/src/OdmMeshing.cpp).  

Poisson surface reconstruction method from PCL is used to create a mesh from the point cloud as seen [here](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_meshing/src/OdmMeshing.cpp#L6)
`meshCreator_ = pcl::Poisson<pcl::PointNormal>::Ptr(new pcl::Poisson<pcl::PointNormal>());`

By my understanding, the Poisson method gives a "watertight" surface and I've been wondering how to get rid of the unwanted faces (& vertices) generated. Is that done at the mesh decimation stage?

Seen [here](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_meshing/src/OdmMeshing.cpp#L322): `pcl::MeshQuadricDecimationVTK decimator`

I'd appreciate any insight to help me better understand the ODM workflow. 

Thanks!

dakotabenjamin, 2016-07-15:
Are you asking about cleaning up the mesh during the process? 

Mesh decimation simplifies over-tessellated parts of the mesh, i.e. a flat face with 50 triangles down to 5 or something like that. Mesh decimation happens only if the generated vertex count is higher than the input max vertex count. The algorithm used is first described here: 
> H. Hoppe. New Quadric Metric for Simplifying Meshes with Appearance Attributes. IEEE
Visualization 1999, pp. 59-66. 

To be quite honest, our meshing process is quite simple and is on our list of things to improve, in terms of both in mesh density and accuracy.

v4ven27, 2016-07-20:
Thank you, @dakotabenjamin 

However, what I am most concerned about is how ODM handles the "watertightness" property of Poisson reconstruction due to which it results in closed surfaces as seen here:
[image]

This is the front view of a building whose surfaces are clearly discernible but you can also see how Poisson Reconstruction has closed the the entire mesh by generating polygons from one stairwell to the other. I'm curious about how ODM handles this issue. :)

Thanks!

dakotabenjamin, 2016-07-20:
Unfortunately, ODM handles it pretty poorly as you can see. 

I would say it will be hard to create a proper solution to photosets with non-nadir facing photos like above. I would avoid using photos with any part of the sky in them at all. I think that is what is causing your issues. 

One thing you can try is use the `--end-with pmvs` tag to stop after pmvs pointcloud densification step. Then manually remove any points rendered as sky or points you don't want by hand using MeshLab or some point cloud editor, then replace the ply and rerun from `odm_meshing`. *Theoretically* that could work.

smathermather, 2016-07-21:
Meshing is definitely something we want to revisit and improve in the near term.

v4ven27, 2016-07-21:
@dakotabenjamin and @smathermather , thanks for your response! I'm currently working with PCL (used in generating the above mesh) and CGAL to process aerial LIDAR pointclouds and generate meshes from non-nadir viewpoints. I'd glanced over the ODM source to see how the watertightness property of Poisson reconstruction is handled in ODM. 

Considering my research area overlaps with ODM as far as the meshing module is concerned, I'd love to contribute towards improving mesh generation in ODM :). Let me know how I can help. 

I'll update this post once I have solid (pun intended) results from my research.

dakotabenjamin, 2016-07-22:
That sounds great! We are welcoming to any and all contributions :smile: If you want to contribute but need some direction or help navigating the codebase, head on over to our [Gitter](https://gitter.im/OpenDroneMap/OpenDroneMap) for dev talk.

closed 2018-03-28 by dakotabenjamin