OpenDroneMap/ODM #339: Code 256 error quitting clause

Bavar2142, 2016-07-11:
Hello all,
ran a small roughly 20 image dataset and had the following error. Was it because the data set was too small?

Traceback (most recent call last):
  File "/home/coonamble/OpenDroneMap/src/OpenSfM/bin/mesh", line 23, in <module>
    reconstructions = data.load_reconstruction()
  File "/home/coonamble/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 291, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: 'opensfm/reconstruction.json'
b

quitting cause: 
	PYTHONPATH=/home/coonamble/OpenDroneMap/lib/python2.7/dist-packages "/home/coonamble/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

dakotabenjamin, 2016-07-12:
That is usually the case. 

How much overlap are these photos?

Bavar2142, 2016-07-12:
About 20 to 30%. That was just a trial run though adnd was a part of a single track. V2 as it were will be at a lower altitude in a grid.

jamiephillips0000, 2016-07-14:
Hi 
I am having the same issue 
I have tried
1) Two very similar images
2) The same image twice
Same error
images attached

[image]
[image]

`

[INFO]    Initializing OpenDroneMap app - Thu Jul 14 09:10:44  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /home/ubuntu/TEMP/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded G0021343.JPG | camera: gopro hero4 silver | dimensions: 4000 x 3000 | focal: 3.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded G0021342.JPG | camera: gopro hero4 silver | dimensions: 4000 x 3000 | focal: 3.0 | ccd: None
[INFO]    Found 2 usable images
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM Resize Cell
[DEBUG]   Resizing dataset to: /home/ubuntu/TEMP/images_resize
[DEBUG]   Resized G0021343.JPG | dimensions: (768, 1024, 3)
[DEBUG]   Resized G0021342.JPG | dimensions: (768, 1024, 3)
[INFO]    Resized 2 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   running PYTHONPATH=/home/ubuntu/git/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/ubuntu/git/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/ubuntu/TEMP/opensfm
2016-07-14 09:10:45,481 Extracting focal lengths for image G0021343.JPG
2016-07-14 09:10:45,500 Extracting focal lengths for image G0021342.JPG
INFO:__main__:Extracting ROOT_HAHOG features for image G0021343.JPG
INFO:__main__:Extracting ROOT_HAHOG features for image G0021342.JPG
DEBUG:opensfm.features:Found 4862 points in 1.44987201691s
DEBUG:opensfm.features:Found 4862 points in 1.45249009132s
Matching 1 image pairs
Matching G0021343.JPG  -  1 / 2
Matching G0021342.JPG  -  2 / 2
G0021342.JPG - G0021343.JPG has 4862 candidate matches
Robust matching time : 0.0281529426575s
Full matching 4725 / 4862, time: 1.30093002319s
2016-07-14 09:10:49,584 reading features
2016-07-14 09:10:49,608 Merging features onto tracks
2016-07-14 09:10:49,653 Good tracks: 4725
images 2
nonfisheye images 2
0 partial reconstructions in total.
Traceback (most recent call last):
  File "/home/ubuntu/git/OpenDroneMap/SuperBuild/src/opensfm/bin/mesh", line 23, in <module>
    reconstructions = data.load_reconstruction()
  File "/home/ubuntu/git/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 291, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/home/ubuntu/TEMP/opensfm/reconstruction.json'
[ERROR]   quitting cause: 
	PYTHONPATH=/home/ubuntu/git/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/ubuntu/git/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/ubuntu/TEMP/opensfm
returned with code 256.

An error occurred. Check stdout above or the logs.

dakotabenjamin, 2016-07-15:
I don't know what the absolute minimum number of images you can have to run ODM (at the recommended 60-75% overlap) but 2 images is too few. Even if you get a reconstruction in OpenSfM with 2 images, you would have to change
```
--pmvs-minImageNum <positive integer>
                      Each 3D point must be visible in at least minImageNum
                      images for being reconstructed. 3 is suggested in
                      general.
```
to 2 to generate a dense point cloud. 

My minumum recommendation is 50 images, given 60% overlap minimum. You may still run into problems though.

Fi156, 2016-07-15:
And you really need to take different images. ;)

Bavar2142, 2016-07-21:
The rerun worked beautifully with 100 is images.

closed 2016-07-21 by Bavar2142