OpenDroneMap/ODM #340: ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

bryanibit, 2016-07-15:
After compiling(bash ./configure.sh), I use command like "cd ./tests/test_data    sudo python ../../run.py" . The error show like following:

Traceback (most recent call last):
  File "../../run.py", line 8, in <module>
    import ecto
  File "/home/inin/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

Of course, firstly,I add "export PYTHONPATH=$PYTHONPATH:~/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/OpenDroneMap/SuperBuild/install/lib" to .bashrc doc. Secondly, I delete something(after #) and add something in configure.sh doc like following:
                     #libboost-filesystem-dev \
                     #libboost-iostreams-dev \
		      #libboost-regex-dev \
                     #libboost-python-dev \
                     #libboost-date-time-dev \
                     #libboost-thread-dev -y -qq
		     libboost-filesystem1.55-dev \
                     libboost-iostreams1.55-dev \
                     libboost-regex1.55-dev \
                     libboost-python1.55-dev \
                     libboost-date-time1.55-dev \
                      libboost-thread1.55-dev -y
Finally, I am not sure whether I modify  self.make_model = None to self.make_model ='' in opendm/types.py doc. So I remain still and didn't modify the doc.
Could you please tell me how to solve the problem? Thank you so much.

dakotabenjamin, 2016-07-15:
We have solved this problem over email. :+1:  Please reopen if more problems.

closed 2016-07-15 by dakotabenjamin

bryanibit, 2016-07-15:
Thank to the patient reply of **dakotabenjamin**, the problem has solved. Let me explain briefly for later one who meets the same problem.  1>When you modify the basrc file, you need to log out and back in again.  2>As a test in OpenDroneMap file, try "bash run.sh --project-path ./tests/test_data" At the same time, close something you not need when running.