OpenDroneMap/ODM #349: Error running configure.sh

AndrewToole, 2016-07-30:
I am following the steps in the readme:
did the git clone
export PYTHONPATH
export LD_LIBRARY_PATH
bash configure.sh
runs for a long time then fails with the message below. It appears there is a NumPy error so I tried a pip install NumPy but it is already installed as version 1.11.0 
I then upgraded numpy to 1.11.1 but it still fails.
Any suggestions would be greatly appreciated.

.......
[ 41%] Performing configure step for 'opengv'
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python
CMake Error at modules/FindNumPy.cmake:62 (message):
  NumPy import failure:

  Traceback (most recent call last):

    File "<string>", line 1, in <module>

  ImportError: No module named numpy

Call Stack (most recent call first):
  python/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/andrew/OpenDroneMap/SuperBuild/build/opengv/CMakeFiles/CMakeOutput.log".
CMakeFiles/opengv.dir/build.make:108: recipe for target 'opengv/stamp/opengv-configure' failed
make[2]: *** [opengv/stamp/opengv-configure] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/opengv.dir/all' failed
make[1]: *** [CMakeFiles/opengv.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Script finished

dakotabenjamin, 2016-08-01:
1. What does your setup look like?
2. Have you tried opening the python console and running `import numpy`?

AndrewToole, 2016-08-01:
I am running Ubuntu 16.04 on VirtualBox 5.1.0
I set up a virtual environment prior to running as I also use python 3 for some other stuff. I wondered if this was the issue so dropped the virtual environment and re-ran with same results.
When I open Python 2.7 numpy imports fine and runs fine
Could it be a 16.04 issue?

dakotabenjamin, 2016-08-01:
I don't think so. Maybe the default pip and the default python for ODM aren't matching up?

AndrewToole, 2016-08-01:
my default python is 2.7.12 and default pip is 8.1.2
I believe ODM is looking for 2.7 not 3 correct?

AndrewToole, 2016-08-02:
I believe I now have it working. When I scrolled way up in the install there was a message about permissions.
I switched to root user and tried the install again and it seems to have worked.

dakotabenjamin, 2016-08-02:
Yes, ODM uses 2.7

I'm glad to see that you are able to install as root, however you ought to be able to do so only with a sudo-accessible account.

closed 2018-03-28 by dakotabenjamin