OpenDroneMap/ODM #350: Distorted results when running ODM using imagery from a Phantom 4

dbaldwin, 2016-08-01:
I've been trying to process my imagery for the past week using the latest version of ODM and imagery from my Phantom 4. This imagery works well using Photoscan Pro, but I'd really like to do a comparison with ODM output. For starters here is my imagery:

https://www.dropbox.com/sh/qd5n9dz9jd2s5u2/AAAJR9D3EpWU8MBfYkOicbZDa?dl=0

There are 95 georeferenced photos and the Photoscan output file. Here is a direct download link to the Photoscan geotif:

https://www.dropbox.com/s/ljnlz29cufs4xr2/tiger_stadium_053016.tif?dl=0

Let me start from the beginning. Trying to run ODM without modifying any of the exif data yielded the following error:

http://lists.osgeo.org/pipermail//opendronemap-users/2016-June/000321.html

That division by zero error came from the EXIF data having a focal length of 0. I'm attaching the exif output before I modified it. I believe this is a bug in DJI's firmware that causes the focallength to be 0. Note that I did not modify any data for Photoscan to run properly.

[dji_phantom4_exif_data.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/392732/dji_phantom4_exif_data.txt)

After digging around a bit I found that the ccd width was not in the sensor_data.json file. So I appended the following:

"DJI FC330": 6.17

Then I proceeded to add the focal length and focal length in 35mm tags to my images like this:

exiftool -FocalLength="3.5 mm" /home/vagrant/OpenDroneMap/images/*.JPG

exiftool -FocalLengthIn35mmFormat="20 mm" /home/vagrant/OpenDroneMap/images/*.JPG

After that I ran ODM again with the following command:

./run.sh --project-path=/home/vagrant/OpenDroneMap --rerun-all

Everything completed but I ended up with a severely distorted ortho that you can see here:

[image]

and here is a screen grab of my point cloud using the web viewer (which is an awesome feature btw):

[image]

You can see that the camera shots are evenly spaced but the point cloud appears to be all over the place. I'm not really sure where to go from here. I'm going to see if I can spend some more time looking through the OpenSFM code and come up with any theories. 

Any insight would be greatly appreciated.

Thanks.

dakotabenjamin, 2016-08-01:
@paulinus have you seen this before? Do you know how to troubleshoot?

dbaldwin, 2016-08-03:
Let me know if I can come up with any other info to help troubleshoot. My bet is it has something to do with the EXIF data. I'm going to try and do a shoot with my Inspire 1 and compare the EXIF data to the P4.

smathermather, 2016-08-04:
I have a hunch -- is the FC330 a wide angle lens? This might be a good candidate for calibration.

dbaldwin, 2016-08-04:
Yes the FC330 is considered wide angle. Is there a tutorial for calibration? I'll gladly give it a try and see how that impacts the results.

dakotabenjamin, 2016-08-04:
I've used http://www.janeriksolem.net/2014/05/how-to-calibrate-camera-with-opencv-and.html with success

dbaldwin, 2016-08-05:
Thanks @dakotabenjamin I will be giving this a try over the weekend and will share my results.

dakotabenjamin, 2016-08-05:
One thing to keep in mind is that you will also have to copy over the EXIF data

cormal, 2016-08-06:
I'd be interested in your calibration vales for the P4 from OpenCV if you end up running it. Could you post them here?

Thanks!

dbaldwin, 2016-08-08:
Here's where I currently stand. I installed OpenCV on a separate Vagrant machine (Ubuntu Trusty 32) and ran the calibrate.py script:

https://github.com/opencv/opencv/blob/master/samples/python/calibrate.py

I took roughly 8 photos of my checkerboard and repeatedly saw the following error:

processing /vagrant/calibration_images/DJI_0109.JPG... chessboard not found

Here is a sample image: 
[image]

I'm thinking maybe there isn't enough contrast between my grey table and the chessboard. I've also read that it's helpful to have more white boundary around the chessboard. Maybe I have too little.

Here is an image from the OpenCV test data set that runs just fine: 
[image]

I'll take another round of shots tomorrow and post my results.

dakotabenjamin, 2016-08-08:
I got that error too when using a checkerboard on my screen that had no white outline. 

Also I don't know how much it affects but https://github.com/opencv/opencv/blob/master/samples/python/calibrate.py#L48 might need to be adjusted.

dbaldwin, 2016-08-08:
@dakotabenjamin I think you nailed it with that line of code. I didn't even notice that! The chessboard I had was from a print I did a couple years ago when calibrating a Canon camera. I updated the code to:

pattern_size = (8, 6)

I didn't realize that the key was to count the # of internal corners. That's what threw me off. With that being said here is my output:

vagrant@vagrant-ubuntu-trusty-32:~/OpenCV/samples/python$ sudo time python calibrate.py "/vagrant/calibration_images/*.JPG"
libdc1394 error: Failed to initialize libdc1394
processing /vagrant/calibration_images/DJI_0118.JPG... ok
processing /vagrant/calibration_images/DJI_0119.JPG... ok
processing /vagrant/calibration_images/DJI_0120.JPG... ok
processing /vagrant/calibration_images/DJI_0121.JPG... ok
processing /vagrant/calibration_images/DJI_0122.JPG... ok
processing /vagrant/calibration_images/DJI_0123.JPG... chessboard not found
processing /vagrant/calibration_images/DJI_0124.JPG... chessboard not found
processing /vagrant/calibration_images/DJI_0125.JPG... chessboard not found

RMS: 2.25395219742
camera matrix:
 [[  2.37220003e+03   0.00000000e+00   1.97761668e+03]
 [  0.00000000e+00   2.36795653e+03   1.49562498e+03]
 [  0.00000000e+00   0.00000000e+00   1.00000000e+00]]
distortion coefficients:  [ 0.03500402  0.11911294 -0.00290307  0.00116473 -0.58046289]

Undistorted image written to: ./output/DJI_0118_chess.png_undistorted.png
libpng warning: Image width is zero in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
Undistorted image written to: ./output/DJI_0119_chess.png_undistorted.png
libpng warning: Image width is zero in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
Undistorted image written to: ./output/DJI_0120_chess.png_undistorted.png
libpng warning: Image width is zero in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
Undistorted image written to: ./output/DJI_0121_chess.png_undistorted.png
libpng warning: Image width is zero in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
Undistorted image written to: ./output/DJI_0122_chess.png_undistorted.png
libpng warning: Image width is zero in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
560.41user 3.41system 9:23.27elapsed 100%CPU (0avgtext+0avgdata 367480maxresident)k
7584inputs+189248outputs (38major+1090533minor)pagefaults 0swaps

Note that in the last 3 images the chessboard was not detected. Also, for some reason the output undistorted images were empty. Perhaps that's an error with libpng on my VM. I don't want to waste a bunch of time tracking that down at the moment. I'd like to give the coefficients a try. @dakotabenjamin can you let me know how to use those at parameters in ODM? I saw you said something about writing data in the EXIF tags, but I'm not 100% sure how to do that.

Lastly, here is a sample chessboard image shot with the P4: 

[image]

and here is the corner detection of the image above after running the calibrate.py script:

[image]

I'm going to run some more photo calibration tests today, but I'd like to run the coefficients through ODM today.

dakotabenjamin, 2016-08-08:
Have you tried converting to JPG? I'll post my undistort code shortly

dakotabenjamin, 2016-08-08:
https://github.com/dakotabenjamin/CameraCalibration

If running on Linux, then the exif will be copied over. Otherwise I think the program will break ungracefully but the photos will still be produced. Note that this does not account for focal length changes that may occur when undistorting the images.

dakotabenjamin, 2016-08-08:
In the future I'd like to be able to simply supply the camera matrix and distortion parameters to ODM and have it handle the image distortion but this is my solution for now.

dbaldwin, 2016-08-08:
Let me try to write as jpg and see what happens. I'm going to take some more with better lighting and will post those results as well. I'd like to come up with a good repeatable process for others. You mentioned:

"Note that this does not account for focal length changes that may occur when undistorting the images."

Does that require any action by the end user or are you simply making a statement? Appreciate your assistance with this.

dakotabenjamin, 2016-08-08:
I don't know enough about photography to give much assistance here, unfortunately. I don't know what to do to correct the focal length, or if it matters at all. So I made the assumption in the script that the focal length was the same as the originals for the undistorted photos.

dbaldwin, 2016-08-08:
I hear you. I'm heading down unchartered waters myself. I was able to take another set of images and get corners matched in all of them. Still no undistorted images were written from the default calibrate.py script even after trying to output to jpg. I'm not going to worry about that right now. Here is my output:

vagrant@vagrant-ubuntu-trusty-32:~/OpenCV/samples/python$ sudo time python calibrate.py "/vagrant/calibration_images/*.JPG"
libdc1394 error: Failed to initialize libdc1394
processing /vagrant/calibration_images/DJI_0108.JPG... ok
processing /vagrant/calibration_images/DJI_0109.JPG... ok
processing /vagrant/calibration_images/DJI_0110.JPG... ok
processing /vagrant/calibration_images/DJI_0111.JPG... ok
processing /vagrant/calibration_images/DJI_0112.JPG... ok
processing /vagrant/calibration_images/DJI_0113.JPG... ok
processing /vagrant/calibration_images/DJI_0114.JPG... ok
processing /vagrant/calibration_images/DJI_0115.JPG... ok
processing /vagrant/calibration_images/DJI_0116.JPG... ok
processing /vagrant/calibration_images/DJI_0117.JPG... ok
processing /vagrant/calibration_images/DJI_0118.JPG... ok

RMS: 1.50009660911
camera matrix:
 [[  2.36585284e+03   0.00000000e+00   1.97024146e+03]
 [  0.00000000e+00   2.36417864e+03   1.49737745e+03]
 [  0.00000000e+00   0.00000000e+00   1.00000000e+00]]
distortion coefficients:  [  1.81517127e-02   1.34277087e-01  -2.80837892e-04  -6.32672126e-04
  -4.88612822e-01]

Undistorted image written to: ./output/DJI_0108_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0109_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0110_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0111_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0112_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0113_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0114_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0115_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0116_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0117_chess.png_undistorted.jpg
Undistorted image written to: ./output/DJI_0118_chess.png_undistorted.jpg
886.50user 5.12system 14:55.30elapsed 99%CPU (0avgtext+0avgdata 462452maxresident)k

Here is a sample of the corner matching: 
[image]

Now referencing back to the original article that you referenced:

http://www.janeriksolem.net/2014/05/how-to-calibrate-camera-with-opencv-and.html

I was able to take my coefficients and plug them into the undistort script as mentioned in the article. Here is the script:

https://gist.github.com/dbaldwin/21d55b134d71fd66e06dce8b91651a03

The original photo:

[image]

and the undistorted version using the coefficients from above:

[image]

I think I'm pretty happy with these results so I'm going to try and run your code to produce all the images and write the exif data. If that is successful then I'll give them a run through ODM.

dakotabenjamin, 2016-08-09:
Great! Let me know how it works.

dbaldwin, 2016-08-09:
@dakotabenjamin today has been hectic already so I haven't been able to diagnose a problem I'm seeing in your undistort script. I'll look at it later today but let me share what's going on in case you have any quick ideas. First off the calibration script run perfectly. Same results as what I saw from the original script yesterday. Your version of the scripted created the matrix.txt and distortion.txt files as expected. Then I ran the following command and you can see the output. For starters, it's trying to create a directory called \out\ which seems strange to me. Maybe that's intentional but seems like a strange format for a directory in *nix. The other problem is that it during the loop it only appears to do the first loop and I guess it fails gracefully.

vagrant@vagrant-ubuntu-trusty-32:~/CameraCalibration$ python undistort.py --matrix /vagrant/calibration_images/matrix.txt --distortion /vagrant/calibration_images/distortion.txt /home/vagrant/images/*.JPG
libdc1394 error: Failed to initialize libdc1394
Matrix: 
[[  2.36585284e+03   0.00000000e+00   1.97024146e+03]
 [  0.00000000e+00   2.36417864e+03   1.49737745e+03]
 [  0.00000000e+00   0.00000000e+00   1.00000000e+00]]
Distortion: [  1.81517127e-02   1.34277087e-01  -2.80837892e-04  -6.32672126e-04
  -4.88612822e-01]
DEBUG:Starting loop
DEBUG:writing images to \out\)
/home/vagrant/images/DJI_0110.JPG
DEBUG:Undistorting /home/vagrant/images/DJI_0110.JPG . . . 
exiftool -TagsFromFile /home/vagrant/images/DJI_0110.JPG -all:all /home/vagrant/images/DJI_0110.JPG_undistorted

Here is the original file:

[image]

and undistorted version:

[image]

Looks like I need to spend some more time troubleshooting. I thought I was close!

dakotabenjamin, 2016-08-09:
My bad. I need to put OS agnostic paths in.

dakotabenjamin, 2016-08-09:
Alrighty so I'm gonna need to spend some more time testing but the patch below might fix it. 

[Make_paths_OS_agnostic.patch.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/409530/Make_paths_OS_agnostic.patch.txt)

dakotabenjamin, 2016-08-09:
OK I think it have it working now. run `git pull` to update

dbaldwin, 2016-08-09:
Working like a champ, but I think there's a small bug. If you see my sample photos above the undistorted version is really off. From this article:

http://www.janeriksolem.net/2014/05/how-to-calibrate-camera-with-opencv-and.html

It talks about just using the first two values from the distortion coefficients. So the last 3 values will be zero like this:

https://gist.github.com/dbaldwin/21d55b134d71fd66e06dce8b91651a03#file-undistort-py-L10

I see that you're loading all 5 values from the distortion.txt file here:

https://github.com/dakotabenjamin/CameraCalibration/blob/master/undistort.py#L44

Actually it looks like you have it solved on line 45 but commented out:

https://github.com/dakotabenjamin/CameraCalibration/blob/master/undistort.py#L44

I just modified your undistort file to ignore the last 3 distortion coefficients and I have a good set of undistorted images to run through ODM. Fingers crossed and will report back later.

dbaldwin, 2016-08-09:
Progress....

[image]

Not sure where to go from here so if you guys have any suggestions I'll be happy to try and sort through them. @dakotabenjamin perhaps this is an issue with the undistortion script but not changing the focal length? I'll see if I can dig anything else up.

dakotabenjamin, 2016-08-15:
I think I'm out of suggestions, unfortunately. I'm out of the office all week but will keep updated. Maybe @smathermather can help here.

KommandorKeen, 2017-05-29:
I am getting pretty good results with my Phantom 4 with a good set of vertical data, but the odd bit of distortion with the trees etc. Anyone taken this further since last August and do peeps now regularly undistort their P4 images before running through ODM?

dakotabenjamin, 2017-06-07:
Given the mass of other improvements to the toolchain in the past year or so, I think this thread should be closed. @KommandorKeen calibration is currently the solution.

closed 2017-06-07 by dakotabenjamin