OpenDroneMap/ODM #352: XMP error running with docker

sear, 2016-08-03:
Hi, 

 I installed the docker image to use on a mac, then tested with a sample dataset (odm_data_benchmark) which worked fine. After that I tried with our own dataset and i got into this error:
$ docker run -v $IMAGES:/images opendronemap/opendronemap
[INFO]    Initializing OpenDroneMap app - Wed Aug 03 18:54:10  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /images/images
Traceback (most recent call last):
  File "/code/run.py", line 38, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/dataset.py", line 57, in process
    self.params.force_ccd)
  File "/code/opendm/types.py", line 32, in __init__
    self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
  File "/code/opendm/types.py", line 62, in parse_pyexiv2_values
    val = metadata[key].value
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 210, in _get_value
    self._compute_value()
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 200, in _compute_value
    self._value = self._convert_to_python(self._raw_value, self.type[17:])
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 374, in _convert_to_python
    raise NotImplementedError('XMP conversion for type [%s]' % type)
NotImplementedError: XMP conversion for type [seq Integer]

i searched this forum for something similar   but found nothing. One thread suggested to adjust FocalLength in the file using exiftool, which i did and gives the same error. I also tried using --force-ccd 6.17 --force-focal 3.5 and combinations of this two parameters and nothing. 

I don't have the drone model as this dataset was given to me to experiment with this software, but by the exif info i can know the camera model (FC300X). Can anyone point some clue on how this could be fixed?

cheers

Sergio

dakotabenjamin, 2016-08-04:
You get the same error output when not using --force-focal?

Can you upload a sample image?

sear, 2016-08-04:
yes the same result with or without any parameters. Here's a sample image.
[image]

dakotabenjamin, 2016-08-04:
Alright so it's failing in a loop and I don't really know what's going on, but with the following patch you can rerun and determine which file and which tag the error is being thrown on: 
```
From a152767e3d5ddf584ee9bb3a37c9883dba450973 Mon Sep 17 00:00:00 2001
From: Dakota Benjamin <dakota.benjamin@gmail.com>
Date: Thu, 4 Aug 2016 14:03:42 -0400
Subject: [PATCH] add debug line

---
 opendm/types.py | 1 +
 1 file changed, 1 insertion(+)

diff --git a/opendm/types.py b/opendm/types.py
index b560e9d..6ed30f3 100644
--- a/opendm/types.py
+++ b/opendm/types.py
@@ -59,6 +59,7 @@ class ODM_Photo:
             # try/catch tag value due to weird bug in pyexiv2
             # ValueError: invalid literal for int() with base 10: ''
             try:
+                log.ODM_DEBUG("Parsing %s for %s" % (key, _path_file))
                 val = metadata[key].value
                 # parse tag names
                 if key == 'Exif.Image.Make':
--
1.9.1
```

If you've never applied patches before, you can also just add that line manually to opendm/types.py

Then rerun and it will tell you the metadata key and the file that pyexiv2 fails on.

sear, 2016-08-04:
Hi Benjamin, 

   I tried to do what you suggested, but i cant or dont know how to update the docker image after applying the patch that you send. This is what i did:

:~ sergio$ docker run -i -t --entrypoint /bin/bash opendronemap/opendronemap
odm@097c5b168aff:/code$  cd opendm/
odm@097c5b168aff:/code$  vi types.patch (and here i copied the patch)
odm@097c5b168aff:/code/opendm$ patch -b < types.patch
patching file types.py
Hunk #1 succeeded at 59 with fuzz 1.
odm@097c5b168aff:/code/opendm$ exit
exit
:~ sergio$ docker commit -m "patch opendm types.py" -a "Sergio" 3088bd33eaef opendronemap/opendronemap:v2 f7482b69ebb9b947618fe12417f56a8a6694e0df890cad54abe1f83d26148201
"docker commit" requires at least 1 and at most 2 argument(s).

and i cant run the patched image because i cant commit the changes. Im sorry im new to docker and dont know how to update the image.

dakotabenjamin, 2016-08-05:
I think when you edit from inside the docker image(?) then you're all good and can run it again. Editing types.py doesn't require a rebuild or reinstallation.

wekorthals, 2016-08-07:
I'm having the same issue as Sergio.

sear, 2016-08-07:
I found out how to commit changes, if you do the docker ps command the first column is the container ID, this is the container referred by the commit command: 

docker commit -m "patch types.py" -a "Sergio" 7be186d29f3e opendronemap/opendronemap:v3

this now allows the changes to the image to become permanent. After this i ran docker again but this time just drops back to the cli, and does nothing. I guess i did something wrong. I try deleting and redownloading the opendronemap docker image again on monday. Or maybe try to use a linux machine to do the tests. 

Thanks for your kind and quick responses!

Cheers

Sergio

dakotabenjamin, 2016-08-08:
You could clone the repo, apply the changes and build the docker image yourself.

sear, 2016-08-11:
thanks again Benjamin, i think im going with the linux option and if there is any issue now i know that the people in this forum is kind to newbies like myself. 

thanks again, cheers

Sergio.

julia-altenbuchner, 2016-09-02:
I tried the same as @wekorthals described (GoPro Hero 4 with exif metadata added through GeoSetter), getting the same error. Adding the line to types.py gives me this:

[DEBUG]   Loading dataset from: /vagrant_data/Matoto_12_10/images
[DEBUG]   Parsing Exif.Image.ImageDescription for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.Make for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.Model for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.Orientation for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.XResolution for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.YResolution for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.ResolutionUnit for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.Software for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.DateTime for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.YCbCrPositioning for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.ExifTag for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExposureTime for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.FNumber for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExposureProgram for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ISOSpeedRatings for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExifVersion for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.DateTimeOriginal for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.DateTimeDigitized for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ComponentsConfiguration for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.CompressedBitsPerPixel for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ShutterSpeedValue for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ApertureValue for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExposureBiasValue for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.MaxApertureValue for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.SubjectDistance for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.MeteringMode for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.LightSource for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.Flash for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.FocalLength for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.MakerNote for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.FlashpixVersion for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ColorSpace for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.PixelXDimension for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.PixelYDimension for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.InteroperabilityTag for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Iop.InteroperabilityIndex for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Iop.InteroperabilityVersion for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExposureIndex for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.SensingMethod for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.FileSource for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.SceneType for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.CustomRendered for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.ExposureMode for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.WhiteBalance for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.DigitalZoomRatio for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.FocalLengthIn35mmFilm for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.SceneCaptureType for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.GainControl for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.Contrast for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.Saturation for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.Sharpness for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.DeviceSettingDescription for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Photo.SubjectDistanceRange for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.Image.GPSTag for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[DEBUG]   Parsing Exif.GPSInfo.GPSVersionID for /vagrant_data/Matoto_12_10/images/G0091246.JPG
[... cut ...]
    val = metadata[key].value
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 210, in _get_value
    self._compute_value()
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 200, in _compute_value
    self._value = self._convert_to_python(self._raw_value, self.type[17:])
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 374, in _convert_to_python
    raise NotImplementedError('XMP conversion for type [%s]' % type)
NotImplementedError: XMP conversion for type [seq Integer]

dakotabenjamin, 2016-09-12:
@mojodna do you understand what's going wrong here and have any advice on what to do ODM side?

dakotabenjamin, 2016-09-12:
OK I have a fix on the way

closed 2018-03-28 by dakotabenjamin