OpenDroneMap/ODM #353: exif.py line 21, in eval_frac ZeroDivisionError: float division by zero

cormal, 2016-08-05:
Issue as per title.
Testing on two images captured by DJI using EXIF.
DJI FC330 specs updated in sensor_data and detailed
ubuntu 16.04 LTS

Ideas?

Regards

cormal, 2016-08-05:
DJI Does not provide focal length EXIF tag.

closed 2016-08-05 by cormal

dbaldwin, 2016-08-05:
Not sure if it helps you at all, but I talk about how I fixed this in this issue:

https://github.com/OpenDroneMap/OpenDroneMap/issues/350

At least that got me past the error you are seeing, but there's still a problem with distorted results. I'm going to be running the camera calibration today as mentioned in the comments.

cormal, 2016-08-06:
Thanks! I'll definitely keep an eye on your progress. Seems like you might be forging the way for me on this one. I have come across another fail point but Ill leave this thread closed and keep an eye on your #350 instead.

Cheers