OpenDroneMap/ODM #354: Increasing cores increases time for OpenSfM to run

dakotabenjamin, 2016-08-05:
I was doing some benchmarking with a vagrant box to get a better idea of minimum processing/memory needs and I came across a strange phenomenon. As I increased the number of cores, the time it took to run OpenSfM did not decrease, but rather it increased significantly! I quickly plotted the data I was recording in R, you can see below: 
```
> waterbury
  cores   ram     sfm cmvs   pmvs meshing texturing georef  ortho
1     2  2048      NA   NA     NA      NA        NA     NA     NA
2     2  4096 4527.64 9.99 234.03   12.89    106.83 152.89     NA
3     2  8192 4939.79 9.75 247.67   23.24    203.58 166.04     NA
4     4  2048      NA   NA     NA      NA        NA     NA     NA
5     4  4096 4285.10 9.39 143.89   24.11    208.76 167.70     NA
6     4  8192 4239.33 9.12 164.81   24.68    228.53 171.16     NA
7     8  4096 7418.40 9.13 109.91   28.38    241.05 178.02     NA
8     8  8192 7880.79 9.32 112.07   26.44    223.98 171.59     NA
9     8 12288 7796.81 9.26 113.91   23.17    185.17 163.55 228.17
> f <- ggplot(waterbury, aes(cores, sfm))
> f + geom_point(aes(size=(ram/1024))) + labs(title="OpenSfM Cost at Varying Cores", x="Number of Cores", y="OpenSfM Cost (seconds)") + guides(size = guide_legend(title="RAM"))
```
[image]

@paulinus I think you would be interested in this discovery

dakotabenjamin, 2016-08-05:
One thing to note is that I had to use the following params to get it to reconstruct completely: 
```
./run.sh --project-path /vagrant_data/testing_data/waterbury/ --time --min-num-features 8000 --resize-to 1024 --force-ccd 7.76
```

dakotabenjamin, 2016-08-05:
Getting the same on the tests/test_data set as well:
```
> toledo
  cores  ram    sfm cmvs   pmvs meshing texturing georef ortho
1     2 2048 369.93 4.46     NA      NA        NA     NA    NA
2     2 4096 372.27 4.45 518.59   20.73    268.67  19.63 20.45
3     2 8192 369.34 4.39 544.75   19.90    259.13  19.00 20.68
4     4 2048 286.19 4.16     NA      NA        NA     NA    NA
5     4 4096 293.73 4.44 352.08   21.36    275.43  21.00 21.97
6     8 4096 409.97 4.11 289.48   21.95    269.84  20.75 20.71
> g = ggplot(toledo, aes(cores,sfm))
> g + geom_point(aes(size=(ram/1024))) + labs(title="OpenSfM Cost at Varying Cores (Toledo)", x="Number of Cores", y="OpenSfM Cost (seconds)") + guides(size = guide_legend(title="RAM"))
```
[image]

pierotofy, 2016-08-05:
What are the specs on the machine that was used to run the benchmark? Not too familiar with how OpenSFM works at this point, but the drop in performance could be caused by disk I/O.

dakotabenjamin, 2016-08-05:
So it's a 14.04 Virtual machine running on Windows 7: Intel Xeon E5-1650 @ 2.50GHz with 32GB RAM
I adjusted the number of cores/ram according to the table above. 

that's a good point about I/O, I don't know either. 
This is the section that takes the longest in OpenSfM:
```
Adding DSC07658.JPG to the reconstruction
Ceres Solver Report: Iterations: 50, Initial cost: 5.572860e+03, Final cost: 5.496871e+03, Termination: NO_CONVERGENCE
setup/run/teardown 6.03964710236/450.782133818/3.09767699242
Remove 3 outliers
Resection DSC07610.JPG inliers: 236 / 255
-------------------------------------------------------
```

pierotofy, 2016-08-05:
I wonder if this is caused by the fact that the processor really has only 6 cores (12 theads).

http://ark.intel.com/products/64601/Intel-Xeon-Processor-E5-1650-12M-Cache-3_20-GHz-0_0-GTs-Intel-QPI

It's safe to say that the Windows 7 hosts probably uses 1 or 2 cores consistently, so that brings the effective number of free cores to 4. The drop in performance might be due to excessive context switching and the way that the VM effectively uses them.

A good discussion on optimal vcpus allocation for VMs: http://serverfault.com/questions/132441/virtualized-cpu-cores-vs-threads

It would be interesting to see a benchmark with 5 cores and 6. I would expect the 5 cores score to be close to the 4 cores' score, while the 6 cores score should start to spike up.

dakotabenjamin, 2016-08-05:
I just ran 6-cores:
```
cores,ram,sfm,cmvs,pmvs,meshing,texturing,georef,ortho
6,12288,5056.96,9.09,131.57,24.49,198.34,167.31,227.69
```

dakotabenjamin, 2016-08-05:
I've now powered up a 16-core (16 real cores) server and I'll run more tests there. 

I've not done much work with parallelization. Is this a problem we can solve on the software side, or just a general hyperthreading boundary?

dakotabenjamin, 2016-08-05:
I can already tell just watching the OpenSfM stdout that it's working much faster.

dakotabenjamin, 2016-08-08:
@pierotofy It looks to me that 4 cores works best on this machine. I'm going to close this issue and I'll have to get back to benchmarking at a later date, but does the data here help get a better picture of the minimum requirements? 

I would summarize saying any modern CPU (Intel i3 or better for instance,) and 4GB RAM are the minimum requirements for running smaller datasets on ODM. Bigger datasets require more RAM.

closed 2016-08-08 by dakotabenjamin

pierotofy, 2016-08-08:
@dakotabenjamin it does, thanks!

dakotabenjamin, 2016-08-08:
Thank you for your help understanding the virtual memory problem!

dakotabenjamin, 2017-04-24:
@pierotofy do you know how this would work on a server with all virtual processors? or how we should be allocating cores there?

pierotofy, 2017-04-24:
If all CPU resources are virtualized, it's assumed that the virtual machine has been set to use the correct number of physical cores, so the number of running threads should equal the number of available CPUs to optimize usage. If in doubt however, running a benchmark is the only way to tell for sure.