OpenDroneMap/ODM #356: How can I process DJI Phantom 2 vision+ photos

herschmi, 2016-08-10:
Hello!

When I try to process DJI photos, I got the next error message:

herschmi-Lenovo-G510 OpenDroneMap # sh run.sh --project-path /home/herschmi/sfmproba
run.sh: 3: run.sh: Bad substitution
[INFO]    Initializing OpenDroneMap app - Wed Aug 10 09:36:33  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /home/herschmi/sfmproba/images
Traceback (most recent call last):
  File "/home/herschmi/OpenDroneMap/run.py", line 38, in <module>
    plasm.execute(niter=1)
  File "/home/herschmi/OpenDroneMap/scripts/dataset.py", line 57, in process
    self.params.force_ccd)
  File "/home/herschmi/OpenDroneMap/opendm/types.py", line 32, in __init__
    self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
  File "/home/herschmi/OpenDroneMap/opendm/types.py", line 62, in parse_pyexiv2_values
    val = metadata[key].value
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 210, in _get_value
    self._compute_value()
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 204, in _compute_value
    self._value = self._convert_to_python(self._raw_value, self.type)
  File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 374, in _convert_to_python
    raise NotImplementedError('XMP conversion for type [%s]' % type)
NotImplementedError: XMP conversion for type [ResourceRef]

What's the problem?

Thank you!

dakotabenjamin, 2016-08-15:
https://github.com/OpenDroneMap/OpenDroneMap/issues/352 is a similar issue. Can you try the patch from there and post the results?

GCorbel, 2016-10-28:
I have the exact same problem on the master branch with a self made Docker image.

I'm trying to work with an image with this info : 

```
<?xml version="1.0" encoding="UTF-8"?>
<x:xmpmeta xmlns:x="adobe:ns:meta/" x:xmptk="XMP Core 4.4.0-Exiv2">
 <rdf:RDF xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#">
  <rdf:Description rdf:about=""
    xmlns:xmp="http://ns.adobe.com/xap/1.0/"
    xmlns:dc="http://purl.org/dc/elements/1.1/"
    xmlns:drone-dji="http://www.dji.com/drone-dji/1.0/"
    xmlns:aux="http://ns.adobe.com/exif/1.0/aux/"
    xmlns:xmpMM="http://ns.adobe.com/xap/1.0/mm/"
    xmlns:stEvt="http://ns.adobe.com/xap/1.0/sType/ResourceEvent#"
    xmlns:stRef="http://ns.adobe.com/xap/1.0/sType/ResourceRef#"
    xmlns:crs="http://ns.adobe.com/camera-raw-settings/1.0/"
    xmlns:tiff="http://ns.adobe.com/tiff/1.0/"
    xmlns:exif="http://ns.adobe.com/exif/1.0/"
    xmlns:photoshop="http://ns.adobe.com/photoshop/1.0/"
   xmp:ModifyDate="2016-07-06T11:29:48-04:00"
   xmp:CreateDate="2016-07-05T14:51:10"
   xmp:CreatorTool="Adobe Photoshop Lightroom 6.6 (Windows)"
   xmp:MetadataDate="2016-07-06T11:29:48-04:00"
   dc:format="image/jpeg"
   drone-dji:AbsoluteAltitude="+82.70"
   drone-dji:RelativeAltitude="+8.50"
   drone-dji:GimbalRollDegree="+0.00"
   drone-dji:GimbalYawDegree="-8.50"
   drone-dji:GimbalPitchDegree="-90.00"
   drone-dji:FlightRollDegree="-11.30"
   drone-dji:FlightYawDegree="-9.30"
   drone-dji:FlightPitchDegree="+2.60"
   aux:SerialNumber="2014031100"
   aux:LensInfo="207000/10000 207000/10000 0/0 0/0"
   aux:Lens="20.7 mm"
   aux:ApproximateFocusDistance="0/100"
   xmpMM:DocumentID="xmp.did:029c424d-ccd5-9d47-b1d2-304592992500"
   xmpMM:OriginalDocumentID="A847A46AC2A70F1186D984B3EE745051"
   xmpMM:InstanceID="xmp.iid:029c424d-ccd5-9d47-b1d2-304592992500"
   crs:Version="9.6"
   crs:ProcessVersion="6.7"
   crs:WhiteBalance="As Shot"
   crs:AutoWhiteVersion="134348800"
   crs:Temperature="4250"
   crs:Tint="+24"
   crs:Saturation="0"
   crs:Sharpness="25"
   crs:LuminanceSmoothing="0"
   crs:ColorNoiseReduction="25"
   crs:VignetteAmount="0"
   crs:ShadowTint="0"
   crs:RedHue="0"
   crs:RedSaturation="0"
   crs:GreenHue="0"
   crs:GreenSaturation="0"
   crs:BlueHue="0"
   crs:BlueSaturation="0"
   crs:Vibrance="0"
   crs:HueAdjustmentRed="0"
   crs:HueAdjustmentOrange="0"
   crs:HueAdjustmentYellow="0"
   crs:HueAdjustmentGreen="0"
   crs:HueAdjustmentAqua="0"
   crs:HueAdjustmentBlue="0"
   crs:HueAdjustmentPurple="0"
   crs:HueAdjustmentMagenta="0"
   crs:SaturationAdjustmentRed="0"
   crs:SaturationAdjustmentOrange="0"
   crs:SaturationAdjustmentYellow="0"
[... 295 lines cut ...]
ws_stitch_1 | [DEBUG]   Parsing Xmp.xmpMM.History[3]/stEvt:when for /tmp/1477685634/images/DJI_0083.jpg
ws_stitch_1 | [DEBUG]   Parsing Xmp.xmpMM.History[3]/stEvt:softwareAgent for /tmp/1477685634/images/DJI_0083.jpg
ws_stitch_1 | [DEBUG]   Parsing Xmp.xmpMM.History[3]/stEvt:changed for /tmp/1477685634/images/DJI_0083.jpg
ws_stitch_1 | [DEBUG]   Parsing Xmp.xmpMM.DerivedFrom for /tmp/1477685634/images/DJI_0083.jpg
ws_stitch_1 | Traceback (most recent call last):
ws_stitch_1 |   File "/opt/OpenDroneMap/run.py", line 38, in <module>
ws_stitch_1 |     plasm.execute(niter=1)
ws_stitch_1 |   File "/opt/OpenDroneMap/scripts/dataset.py", line 57, in process
ws_stitch_1 |     self.params.force_ccd)
ws_stitch_1 |   File "/opt/OpenDroneMap/opendm/types.py", line 32, in __init__
ws_stitch_1 |     self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
ws_stitch_1 |   File "/opt/OpenDroneMap/opendm/types.py", line 63, in parse_pyexiv2_values
ws_stitch_1 |     val = metadata[key].value
ws_stitch_1 |   File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 210, in _get_value
ws_stitch_1 |     self._compute_value()
ws_stitch_1 |   File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 204, in _compute_value
ws_stitch_1 |     self._value = self._convert_to_python(self._raw_value, self.type)
ws_stitch_1 |   File "/usr/lib/python2.7/dist-packages/pyexiv2/xmp.py", line 374, in _convert_to_python
ws_stitch_1 |     raise NotImplementedError('XMP conversion for type [%s]' % type)
ws_stitch_1 | NotImplementedError: XMP conversion for type [ResourceRef]
```

Would it be better to just ignore this error ? The whole problem fail just because an error we probably don't care.

What you think ?

GCorbel, 2016-10-28:
I don't know if it's easy or not but is possible to change for gexiv2 , pyexiv2 seems to be deprecated.

dakotabenjamin, 2016-10-31:
Right now OpenSfM is dependent on pyexiv2 so it doesn't make sense **yet**. I'll open an issue over there though to hear their opinions on the matter.

paulinus, 2016-12-16:
@dakotabenjamin, OpenSfM uses `pyexiv2` only when converting video to still images, which is a feature not used in ODM. So it is possible to migrate ODM to `gexiv2` independently of OpenSfM.

closed 2020-04-14 by smathermather