OpenDroneMap/ODM #359: about using orthophoto module  individually (dev branches)

assbben, 2016-08-20:
i try to output a jpg file when my inputFile(xxx.obj) was produced by other sfm software
but it didnt work well
here is error information
segmentation fault (core dumped)
after i open core file by gdb
here is information 
#0  0x0000000000472a86 in Eigen::DenseStorage<float, 2, 2, 1, 0>::DenseStorage(Eigen::DenseStorage<float, 2, 2, 1, 0> const&) ()
#1  0x000000000046c62b in Eigen::PlainObjectBase<Eigen::Matrix<float, 2, 1, 0, 2, 1> >::PlainObjectBase(Eigen::PlainObjectBase<Eigen::Matrix<float, 2, 1, 0, 2, 1> > const&) ()
#2  0x0000000000468921 in Eigen::Matrix<float, 2, 1, 0, 2, 1>::Matrix(Eigen::Matrix<float, 2, 1, 0, 2, 1> const&) ()
#3  0x0000000000463336 in OdmOrthoPhoto::loadObjFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pcl::TextureMesh&) ()
#4  0x000000000045d76e in OdmOrthoPhoto::createOrthoPhoto() ()
#5  0x000000000045b588 in OdmOrthoPhoto::run(int, char**) ()
#6  0x000000000048b141 in main ()

and now i have  3 questions

1.whether my texture file is too large(88.9mb) for odm to run
2.whether my laptop memory(8GB) is not big enough
thank you

dakotabenjamin, 2016-09-01:
I'm not sure that this will work, but I would try compiling [OdmOrthophoto](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_orthophoto/) then running that with all the proper arguments (found in parseArgs: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp)

assbben, 2016-09-03:
sorry i forgot to put my command to the issue 

cd OpenDroneMap/build/bin/

./odm_orthophoto -inputFile '/home/xxx/opdm_mytest/odm_texturing/odm_textured_model_geo.obj' -outputFile '/home/xxx/Desktop/test_mvs/test.png' -resolution 20

im sure it can work when i used obj made by odm
when i copy obj obj.mlt file(need this file) and textured file to other document it can work too

closed 2018-03-28 by dakotabenjamin