OpenDroneMap/ODM #361: Build errors Fedora 24, gcc 6

sgielkens, 2016-08-23:
Hello,

Thank you for developing OpenDroneMap. It is a very much appreciated complement to make a complete pipeline. I am not an official Fedora packager but I have created several Fedora RPM packages that are related to photogrammetry. OpenDroneMap is one of them:

https://copr.fedorainfracloud.org/coprs/sgielkens/Photogrammetry/package/OpenDroneMap/

For this package I use only the ODM modules. For the remainder I prefer system packages or at least separate packages to keep track of dependencies and to have a reproducible build.

I encounter some build errors. I tested both version 1.0 and the dev branch (commit 02a52e6 on Aug 9, 2016). I use:

gcc --version
gcc (GCC) 6.1.1 20160621 (Red Hat 6.1.1-3)

Eigen3 version: 3.2.9

Build errors:

**odm_georef**

```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_georef/src/Georef.cpp:1402:60: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
                 mesh.tex_coordinates.push_back (coordinates);
                                                            ^
In file included from /usr/include/c++/6.1.1/vector:64:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointCloud2.h:9,
                 from /usr/include/pcl-1.8/pcl/TextureMesh.h:44,
                 from /usr/include/pcl-1.8/pcl/io/obj_io.h:40,
                 from /builddir/build/BUILD/OpenDroneMap-dev/modules/odm_georef/src/Georef.cpp:2:
/usr/include/c++/6.1.1/bits/stl_vector.h:914:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(const value_type& __x)
       ^~~~~~~~~
/usr/include/c++/6.1.1/bits/stl_vector.h:914:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘const value_type& {aka const std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&}’
/usr/include/c++/6.1.1/bits/stl_vector.h:932:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(value_type&& __x)
       ^~~~~~~~~
/usr/include/c++/6.1.1/bits/stl_vector.h:932:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::value_type&& {aka std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&&}’
```

**Similar here:**

```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_georef/src/Georef.cpp:1449:54: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
         mesh.tex_coordinates.push_back(texcoordinates);

```
         
**odm_orthophoto**

Same errors as in odm_georef:

```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp: In member function ‘void OdmOrthoPhoto::createOrthoPhoto()’:
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:438:58: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
         mesh.tex_coordinates.push_back(textureCoordinates);
```
         
```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:1208:60: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
                 mesh.tex_coordinates.push_back (coordinates);
```
                 
```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:1255:54: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
         mesh.tex_coordinates.push_back(texcoordinates);
```

**odm_texturing**

Also Eigen related but somewhat different:

```
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.cpp: In member function ‘void OdmTexturing::createTextures()’:
/builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.cpp:1018:30: error: no match for ‘operator=’ (operand types are ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >’ and ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1> > >’)
     mesh_->tex_coordinates = textureCoordinatesVector;
                              ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/6.1.1/vector:69:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointField.h:9,
                 from /usr/include/pcl-1.8/pcl/point_traits.h:48,
                 from /usr/include/pcl-1.8/pcl/register_point_struct.h:57,
                 from /usr/include/pcl-1.8/pcl/point_types.h:44,
                 from /builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.hpp:8,
                 from /builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.cpp:1:
/usr/include/c++/6.1.1/bits/vector.tcc:167:5: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >]
     vector<_Tp, _Alloc>::
     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/6.1.1/bits/vector.tcc:167:5: note:   no known conversion for argument 1 from ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1> > >’ to ‘const std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >&’
In file included from /usr/include/c++/6.1.1/vector:64:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointField.h:9,
                 from /usr/include/pcl-1.8/pcl/point_traits.h:48,
                 from /usr/include/pcl-1.8/pcl/register_point_struct.h:57,
                 from /usr/include/pcl-1.8/pcl/point_types.h:44,
                 from /builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.hpp:8,
                 from /builddir/build/BUILD/OpenDroneMap-dev/modules/odm_texturing/src/OdmTexturing.cpp:1:
/usr/include/c++/6.1.1/bits/stl_vector.h:450:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^~~~~~~~
/usr/include/c++/6.1.1/bits/stl_vector.h:450:7: note:   no known conversion for argument 1 from ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1> > >’ to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >&&’
/usr/include/c++/6.1.1/bits/stl_vector.h:471:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >]
       operator=(initializer_list<value_type> __l)
       ^~~~~~~~
/usr/include/c++/6.1.1/bits/stl_vector.h:471:7: note:   no known conversion for argument 1 from ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1> > >’ to ‘std::initializer_list<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >’
```

Atiq-Hashmi, 2016-11-20:
Hello There,
Did you find any solution for above post? I have same issue with odm_texturing(The last one in above errors)

Thanks,
Atiq

SW1-SCW, 2017-03-21:
Hi All,

I am having many of the same issues. Any headway?
Building with:
eigen3, version: 3.3.3
gcc (Solus) 6.3.0

Thanks,
SW1-SCW

closed 2020-04-14 by smathermather