OpenDroneMap/ODM #362: ImportError: No module named csfm

dligocki, 2016-08-23:
I have read through the other issues that match my issue but none of the supplied resolutions seem to work. I have tried setting up on a ubuntu 14.04LTS using the Master and DEV branches and the same on a ubunutu 16.04LTS with the exact same error message in each instance. I have tried with my own data as well as example data from here. Any suggestions or ideas would be greatly appreciated.

Here is the message I get in terminal:

Traceback (most recent call last):
  File "/home/damion/Downloads/odm/OpenDroneMap/SuperBuild/src/opensfm/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/damion/Downloads/odm/OpenDroneMap/SuperBuild/src/opensfm/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
[ERROR]   quitting cause: 
	PYTHONPATH=/home/damion/Downloads/odm/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/damion/Downloads/odm/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/damion/Downloads/odm/odm_data_pacifica/opensfm
returned with code 256.

rionlerm, 2016-09-20:
Hi @dligocki ,

did you figure the issue out?

When I installed the dev branch on Ubuntu 14.04 and performed all procedures as stipulated in the instructions and ran the first time it said

> ...No module named ecto

I realised that I didn't restart the terminal/command line for the `~/.bashrc` edits (as per instructions) to take effect. After restarted termianl and reran `bash configure.sh` from the ODM directory all was fine.

dakotabenjamin, 2016-09-20:
Few questions: do you have OpenCV3 installed? Have you tried rerunning configure.sh? Do you get any error messages during installation related to OpenSfM directly?

jpmaniac, 2016-11-07:
I have the same issue... trying to run in docker and it gives me that error. I'm on Ubuntu 16

Right now, I'm creating a Ubuntu 14.04 VM to try and build it in it's own environment, hoping that is a good fix for the issue

closed 2018-03-28 by dakotabenjamin