OpenDroneMap/ODM #365: when attempting to run open drone map: Permission Denied

Bavar2142, 2016-08-24:
When I attempt to run open drone map in a new install it says permission is denied even when I have sudo access.

Bavar2142, 2016-08-24:
Does anyone have any ideas?

v4ven27, 2016-08-25:
Can you provide more information about your ODM installation?

Bavar2142, 2016-09-11:
Its on 16.04 i believe running via a vm. Does that help?

dakotabenjamin, 2016-09-12:
Does your user have sudo privileges?

Bavar2142, 2016-09-14:
Ill give that a go and let you know if it works or not

Bavar2142, 2016-09-14:
Sadly I get the same error.

dakotabenjamin, 2016-09-14:
Can you post screenshots?

Bavar2142, 2016-09-14:
here we are
[image]

v4ven27, 2016-09-14:
Have you tried running this command ? :
` chmod +x run.py `

Bavar2142, 2016-09-14:
Ill try that now.

Bavar2142, 2016-09-15:
chmod cannot access 'run.py': no such file or directory.

v4ven27, 2016-09-15:
You should run the above command in the OpenDroneMap directory.

Bavar2142, 2016-09-15:
Ok ive done that. What do i do next?

Bavar2142, 2016-09-15:
[image]

dakotabenjamin, 2016-09-15:
Make sure you are following the instructions in the README. If you run into errors, document exactly what part of the instructions you were on, what command you ran, and what errors appeared, with screenshots if possible. I'm working on improving the documentation and would like to know where the instructions are confusing, lacking, or misleading.

Bavar2142, 2016-09-15:
So far my procedure has just been the renanme command then ../OpenDroneMap/run.py. Will attempt readme.

Bavar2142, 2016-09-15:
python run.py --project-path /path/to/project

using the wiki I found this command.

I take it the command should be  "python run.py --Pictures/images"?

smathermather, 2016-09-16:
Hi @Bavar2142, more like:
```python run.py --project-path ~/Pictures/images```

Bavar2142, 2016-09-18:
[image]

Like this?

Bavar2142, 2016-09-18:
Ignore that second command.

Bavar2142, 2016-09-20:
If we cant solve this issue ill revert back to 15.10 or .04 as I dont think i have had issues with either of those two versions.

dakotabenjamin, 2016-09-20:
I don't think this is an issue with your operating system. Actually @smathermather may have given you a misleading command call. What does the file tree look like starting at `~/Pictures`?

Bavar2142, 2016-09-21:
Do you mean above or below? Below its just the images folder itself. Above its just home/gis/Pictures/images

Bavar2142, 2016-09-25:
Thanks for the help everyone. Ill revert back to 15.04 for now.

dakotabenjamin, 2016-09-26:
It looks like you don't have a project folder setup. You will want to create a project folder as follows: Let's say you flew a site in Brooklyn, NY. I would recommend setting up a directory structure like this: `~Pictures/ODMProjects/Brooklyn/` then place the images in an a subdirectory of ./Brooklyn. It will look like this:
```
~
|-- Pictures
    |-- ODMProjects
        |-- Brooklyn
            |-- images
                |-- IMG_0001.JPG
        |-- Waterbury
        |-- . . . 
```

Then you will call ODM as such:
```
python run.py --project-path ~/Pictures/ODMProjects/Brooklyn
```

Bavar2142, 2016-09-28:
Upon attempting that line of code this was the result.
odm@odm-VirtualBox:~$ python run.py --project-path ~/Pictures/ODMProjects/Example/images
python: can't open file 'run.py': [Errno 2] No such file or directory

Bavar2142, 2016-09-28:
I take it I run that straight into the terminal without cd'ing anywhere?

dakotabenjamin, 2016-09-28:
Make sure you are referencing run.py where it is.

Bavar2142, 2016-09-28:
The run.py is in the ODM folder. Does that change the code I typed in above?

dakotabenjamin, 2016-09-29:
Yes, if you type `python run.py` it looks in the current directory (in your case above, `~` or `/home/odm/`). You either need to `cd` into the ODM folder or specifically reference run.py (~/ODM/run.py) or else python won't know where to look.

Bavar2142, 2016-09-30:
Hey @dakotabenjamin thanks for the help so far. This is what I get when I run python run.py from the ODM folder.

odm@odm-VirtualBox:~/OpenDroneMap$ python run.py
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/odm/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

dakotabenjamin, 2016-10-03:
Check that you set your environment variables correctly: https://github.com/OpenDroneMap/OpenDroneMap#steps-to-get-opendronemap-running

dakotabenjamin, 2016-10-03:
Or trade `python run.py` with `./run.sh` which sets them for you.

Bavar2142, 2016-10-03:
[image]

is what I get.

dakotabenjamin, 2016-10-11:
What i mean is do this:
odm@odm-VirtualBox:~$ ./run.sh --project-path ~/Pictures/ODMProjects/Example/images

Bavar2142, 2016-10-11:
I run that line of code both outside the odm directory "no such file" and within "you must place your files in an image directory". Any idea as to what im doing wrong?

dakotabenjamin, 2016-10-12:
Where are the images again?

Bavar2142, 2016-10-12:
in an images folder within the pictures folder.

dakotabenjamin, 2016-10-13:
from within the odm directory: `./run.sh --project-path ~/Pictures/`

I think this is helping me realize that the way we ask users to set up their mapping projects is confusing and should be changed for a better user experience.

Bavar2142, 2016-10-13:
I'd have to agree with you there. From an end users perspective something like node ODM or Web ODM is preferred as its basically "click and drag the pictures click go and wait" and the fewer stages something has the fewer chances users have to make a mistake. Im rather excited to be able to get my hands on web ODM once it starts kicking.

Perhaps I should clarify the location of the images directory. It is within the pictures folder of the home directory.

dakotabenjamin, 2016-10-17:
You should make your folders look like https://github.com/OpenDroneMap/OpenDroneMap/issues/365#issuecomment-249680835

closed 2016-10-25 by Bavar2142