OpenDroneMap/ODM #366: Handling of photo exposure differences in orthophoto

rionlerm, 2016-08-26:
Hi all,

is/are there parameters to set that will blend photo exposure differences better in ODM orthophoto creation (see image below)?

It is important to eliminate such anomalies as it conflicts with accurate habitat/land cover classification in a GIS.

[image]

dakotabenjamin, 2016-08-27:
MVSTexturing has been introduced to the dev branch.  

It handles seam leveling, photo consistency, and color correction. Check it out and see if that works for you.

rionlerm, 2016-09-13:
@dakotabenjamin , I suppose the `bash configure.sh` will update the dev branch from the ODM directory to see the effects of the updated branch?

rionlerm, 2016-09-13:
I did the above and followed dev branch install instructions but, seams and colour correction didn't seem to improve?:

[image]

dakotabenjamin, 2016-09-14:
Odd. Can you give some more details about your upgrade process?

rionlerm, 2016-09-14:
I added the 3 export lines to the ~/.bashrc and then performed bash
configure .sh from ODM folder. Should I reclone the ODM GitHub repository?

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2016-09-14:
Did you run `git checkout dev` first?

rionlerm, 2016-09-14:
No I didn't. Sorry but, before *bash configure.sh*?

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2016-09-14:
[image: Inline images 1]

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2016-09-14:
Yes, so:
```
git checkout dev
bash configure.sh
```

You don't need to edit the .bashrc file again though.

rionlerm, 2016-09-14:
Did you see my screenshot?

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2016-09-14:
No, you have to post it on GitHub directly. It won't work over email for some reason.

rionlerm, 2016-09-14:
[image]

dakotabenjamin, 2016-09-14:
Is that directory a git repository?

rionlerm, 2016-09-14:
The ODM folder where ODM was installed to and used to run from.

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2016-09-14:
If it's not a git repo then I assume you downloaded a zip file to install ODM. You can either 
(a) clone the repo and then run `git checkout dev` to checkout the dev branch, or
(b) download the dev zip: https://github.com/OpenDroneMap/OpenDroneMap/archive/dev.zip

rionlerm, 2016-09-14:
Maybe I should reinstall everything?

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2016-09-14:
> archive/dev.zip

Thanks!

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

Bavar2142, 2016-09-14:
@dakotabenjamin 
How would I go about updating to the dev branch? I have been having that same issue with some of our latest mosaics?

dakotabenjamin, 2016-09-15:
I would recommend reading up on how git works. It's a really useful tool but can take some time to understand. Basically once you switch to the dev branch (which you did sucessfully) you need to reinstall using the instructions from the dev branch's [README](https://github.com/OpenDroneMap/OpenDroneMap/blob/dev/README.md). Then you can run normally.

rionlerm, 2016-09-15:
Cool,

busy with that. Thanks.

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2016-09-20:
Huge improvement in dev branch. Well done to all:

rionlerm, 2016-09-20:
Well done to all with dev branch thus far. The result looks much better!

[image]

closed 2016-09-20 by rionlerm

dakotabenjamin, 2016-09-20:
That does look good!