OpenDroneMap/ODM #367: No issue, i create an gcp_list.txt generator i want to share

wolkstein, 2016-08-27:
Hi,
i create an gcp-list.txt generator which automatic find GCP's in an Image Data Set and create  an the right gcp-list.txt file. you only need a koords.txt file which contains your GCP's Coordinates sorted by number and corresponding GCP_Template images from current Data Set.

Example template:
[image]

Example koords.txt
```
WGS84 UTM 39N
P1 631625.6784502362 964921.7122958871 100
P2 631629.4935495785 964967.6482094 100
P3 631696.5718546504,964995.505845105 100
P4 631699.0934857756 964894.2005365156 100
P5 631687.8171783998 964928.6878246851 100
```

More info at  https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator
/g
wolke

dakotabenjamin, 2016-10-03:
Thanks! I'll remember to add this somewhere when I update the documentation.

smathermather, 2020-04-14:
I forgot about this one. Cool tool.

pierotofy, 2020-09-11:
It would be really cool to polish / integrate this in the OpenDroneMap organization. @wolkstein would you be interested (I know it's been 4 years, so not sure if this is still active)?

wolkstein, 2020-09-13:
at the moment i have unfortunately no free time to integrate that. sorry. i had also thought that there are now similar tools within ODM for this.

smathermather, 2020-09-13:
We definitely understand having no free time. There definitely isn't an equivalent tool in the toolkit, so hopefully someone can find some time to integrate your work in the future.

closed 2022-07-26 by pierotofy