OpenDroneMap/ODM #369: [ERROR]   You must put your pictures into an <images> directory

jpwhitney, 2016-08-31:
Following the directions here, https://raw.githubusercontent.com/wiki/OpenDroneMap/OpenDroneMap/Docker.md, everything runs smoothly until I run:`docker run -it --user root\
         -v $(pwd)/images:/code/images\
         -v $(pwd)/odm_orthophoto:/code/odm_orthophoto\
         -v $(pwd)/odm_texturing:/code/odm_texturing\
         --rm odm_image`

At which time I get the following output: `[INFO]    Initializing OpenDroneMap app - Wed Aug 31 07:04:49  2016
[INFO]    Running ODM Load Dataset Cell
[ERROR]   You must put your pictures into an <images> directory
[INFO]    OpenDroneMap app finished - Wed Aug 31 07:04:49  2016
`

I tried putting images into the $(pwd)/images/ directory, but still I get the same error.

Am I running the code wrong or is this a bug?

dakotabenjamin, 2016-09-01:
Is this the dev branch?

jpwhitney, 2016-09-06:
I don't know.  Does the following code checkout the dev branch?
`    git clone https://github.com/OpenDroneMap/OpenDroneMap.git
    cd OpenDroneMap
    docker build -t packages -f packages.Dockerfile .
    docker build -t odm_image .
    docker run -it --user root\
         -v $(pwd)/images:/code/images\
         -v $(pwd)/odm_orthophoto:/code/odm_orthophoto\
         -v $(pwd)/odm_texturing:/code/odm_texturing\
         --rm odm_image`
Code is from here, https://raw.githubusercontent.com/wiki/OpenDroneMap/OpenDroneMap/Docker.md.

dakotabenjamin, 2016-09-06:
OK try this instead:
```
cd OpenDroneMap

git checkout dev

docker build -t packages -f packages.Dockerfile .

docker build -t odm_image .

docker run -it --user root\
     -v $(pwd)/images:/code/images\
     -v $(pwd)/odm_orthophoto:/code/odm_orthophoto\
     -v $(pwd)/odm_texturing:/code/odm_texturing\
     --rm odm_image

```

Marwe, 2016-10-10:
Something seems to be wrong with the entrypoint in the :latest image

    $docker inspect opendronemap/opendronemap
        "RepoTags": [
            "opendronemap/opendronemap:latest"
        ],
            "Cmd": [
                "/bin/sh",
                "-c",
                "#(nop) ENTRYPOINT [\"python\" \"/code/run.py\" \"--project-path\" \"/images\"]"
            ],

if you override the entrypoint with bash:

    docker run -it --user root\
     -v $(pwd)/images:/code/images\
     -v $(pwd)/odm_orthophoto:/code/odm_orthophoto\
     -v $(pwd)/odm_texturing:/code/odm_texturing\
    --entrypoint=/bin/bash\
     --rm odm_image

then you see, that the working dir is /code and  --project-path should be ., not images/ (error exists, because there is no dir images/images/). So, from the bash in the container you can run `python /code/run.py --project-path .`

dakotabenjamin, 2016-10-11:
I think you may be conflating the dev branch with the 'latest' tag (which is pulled from master). There are some important docker changes between dev and master.

kikislater, 2016-10-16:
Confirmed ! 
docker latest does not work now. It's a little bit confusing that we have to use dev branch ...

kikislater, 2016-10-16:
Confirmed ! 
docker latest does not work now. It's a little bit confusing that we have to use dev branch ...

Disk usage = 5.1go after building odm_image

dakotabenjamin, 2016-10-18:
I will get to this as soon as I can

closed 2018-03-28 by dakotabenjamin