OpenDroneMap/ODM #371: Order of cameras in image_list.txt and textured mesh quality improvement

v4ven27, 2016-09-07:
Hello ODMers, 

I was messing around with the odm_texturing module when I came across something bizarre.

I'm using 45 images for textured mesh generation. I noticed that the [image_list.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/460288/image_list.txt) doesn't have any particular ordering for the cameras. Also, I am aware that the camera poses are represented in the `bundler.out` file in the same order as cameras in `image_list.txt`. 

Upon importing the `bundle.out` and `image_list.txt` in the [loadCameras](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_texturing/src/OdmTexturing.cpp#L269) method in OdmTexturing.cpp, we get the camera poses and corresponding texture filenames for each `pcl::texture_mapping::Camera` object which are then pushed onto `cameras_` (which is a `pcl::texture_mapping::CameraVector` object).

I was curious about how changing the order of elements in `cameras_`  will affect texture mapping. 

Here's an screenshot of the textured mesh generated by using the vector as is:

[image]

Here's an screenshot of the textured mesh generated swapping the first four elements of `cameras_`  with the last four elements:

[image]

Now, below is the screenshot of the textured mesh generated after sorting `cameras_`  using std::sort with a functor to compare the  `texture_file` string of each camera object:

[image]

As can be seen, for the same set of inputs (and default arguments to run.sh), the textured mesh looks significantly different when I modify the order of the elements in the `cameras_` vector. The sorted `cameras_` vector (Figure 3) looks best. 

You can find my modified OdmTexturing.cpp [here](http://pastebin.com/Jiudb8Mu). The key changes are in lines 432- 437, in the `triangleToImageAssignment` method.

I welcome any thoughts on the following:
- Is this behavior expected? 
- On what basis does Bundler decide the camera order? 

Thanks!

v4ven27, 2016-09-07:
Here are my observations with the tests/test_data images provided with ODM:

With unsorted `camera_`: 
[image]

Without sorted `camera_`:
[image]

It can be seen that the tree on the bottom left is more clear in the second image, and a parked car is visible under the tree.

smathermather, 2016-09-09:
@v4ven27 -- two (related) questions: which branch are you using that uses bundler? OpenSfM should be the SfM engine its using (although perhaps with bundler formats). Also, have you tried the dev branch? The texturing engine has been replaced and substantially improved using an existing library:
https://www.youtube.com/embed/Ie-qLJdmlLI

v4ven27, 2016-09-09:
@smathermather I'm currently using the main branch which uses OpenSfM (which combines Bundler+PMVS). I'll checkout the dev branch.. thanks!

dakotabenjamin, 2016-09-12:
@v4ven27 if you see the same issue in the dev branch we welcome pull requests!

closed 2018-03-28 by dakotabenjamin