OpenDroneMap/ODM #378: Is there an image processing status bar/indicator during processing?

rextan92, 2016-09-20:
I would like to know whether there is a status bar/indicator telling me that the estimated time required for the processing to be done. Thank you.

Bavar2142, 2016-09-20:
From memory I just go by a rough knowledge of the stages rather than a concrete progress bar.

rextan92, 2016-09-20:
May I know what are the stages and how do they look like?

dakotabenjamin, 2016-09-20:
Resize - OpenSfM - CMVS - PMVS - Meshing - Texturing - Georeferencing - Orthophoto

By and large the stages that take the longest are PMVS followed by OpenSfM and Texturing. You can also run with the `--time` arg to see at the end a rough approximation of the time it took to run each module.

Bavar2142, 2016-09-21:
@dakotabenjamin Is it a goal to eventually have some kind of progress bar? Whilst not important to the program itself it would be a nice quality of life feature.

pierotofy, 2016-09-22:
+1

This would be beneficial for `WebODM` as well. I thought of looking at the console output to infer the progress from the presence of certain keywords (but that could break). It could be as simple as outputting "Progress: (number 0 <= n <= 100)".

Bavar2142, 2016-09-25:
WebODM looks like its be a life saver for me and my workplace. A GUI would really help with the ease of use of the program. @pierotofy

pierotofy, 2016-09-25:
@Bavar2142 agreed. WebODM will probably reach early beta status sometimes in 2017. For a barebone GUI, check out [node-OpenDroneMap](http://github.com/pierotofy/node-OpenDroneMap).

Bavar2142, 2016-09-25:
Thanks ill check that out. Currently trying to get ODM to work on 16.04 to no avail so ill revert back to 15.04.

Bavar2142, 2016-09-25:
@pierotofy Am i right in thinking its browser based in some fashion or another?

Bavar2142, 2016-09-28:
Whilst node may not have a progress bar it is a nice "upgrade" to ODM in terms of ease of use @rextan92 . You may need to tweak resize-to and/or the min-num-features options.

dokutagero, 2017-02-08:
Is this something that might be relevant @dakotabenjamin ? Currently I'm studying a little bit the code base of ODM in order to feel confortable when contributing. So far this would be my first open source project I contribute to but I'm really interested. If this is still a thing that might be useful I could work on it using progressbar2 since I used it in other python projects.

dakotabenjamin, 2017-02-08:
It's something we want but I don't know the best way to go about it, given the difficulties predicting/tracking actual progress of each module.

pierotofy, 2017-02-08:
@dokutagero it would be awesome to add a progressbar feature.

A starting point could be to assign a weight to each module (an estimate of how time intensive each module on a scale from 0 to 1, with all weights adding up to 1), then pick a main loop in each module and calculate how far in the loop you are, multiply by the weight and keep a sum of the progress.

smathermather, 2017-02-08:
That's a great way to keep it simple @pierotofy.

pierotofy, 2020-02-03:
We have rough progress bars as part of NodeODM (and subsequently WebODM). Can this be closed then?

closed 2020-02-04 by smathermather