OpenDroneMap/ODM #380: DSM artefacts

rionlerm, 2016-09-20:
Hi,

I don't know whether it's QGIS but, when creating a DSM with las2dem (LAStools: https://rapidlasso.com/lastools/) and using ODM's ply.las products the following artefacts can be seen. This issues has been noticed before but, doesn't show on all generated DSMs:

[image]

Any ideas?

(PS apart from the artefacts the result is impressive)

dakotabenjamin, 2016-09-20:
I've not used las2dem, can you explain the artefacts to me? Are they the little blue lines?

rionlerm, 2016-09-20:
Correct, small blue lines. I've been using LAStools constantly. The only
thing that changed was ODM.

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2016-09-20:
I tried this: `las2dem -i odm_georeferenced_model.ply.las -o dem1.tif`

and got: 
[image]

rionlerm, 2016-09-21:
Thanks,

it seems your step size remained at 1.0. Changing to 0.01 produces a much
finer DSM.

Any way, I don't find the artefact problem with all DSMs but, will try your
command line paras'.

Regards,
*Rion Lerm (jnr.)*
(+27) 76 913 8381 (Cell)
(+27) 13 735 3536 (Office)
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)

Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

smathermather, 2016-09-25:
@rionlerm -- are you using a licensed lastools? If not, it will introduce regular artifacts like this.

rionlerm, 2016-09-26:
Hello @smathermather ,

I don't use a licensed version but, these artefacts are NOT regular. Most of the DSMs I generate with LAStools show none.

smathermather, 2016-10-04:
I don't think it does it every time... . You might try doing it in https://PDAL.io and see if you get the same artifacts.

closed 2018-03-28 by dakotabenjamin