OpenDroneMap/ODM #383: Orthos are rotated 180º when ignored photos are present in input images directory

mojodna, 2016-09-25:
I started with a folder full of images that didn't include geotags and used `exiftool` to add appropriate location tags to them. `exiftool` helpfully created `<img>.JPG_original` backups of each image. The early stages of ODM reads the location tags appropriate (ignoring anything where the filename does not appear to be an image).

However, sometime during `odm_georeferencing` the location tags are re-read (presumably to create a transformation matrix). My guess is that all files in the directory are read and a matrix that effects a 180º rotation is created.

Deleting the `<img>.JPG_original` files from the input direction eliminates the rotation.

dakotabenjamin, 2016-09-26:
odm_georeferencing calls extract_utm with a -imageslistfile param. What is the contents of that file?

mojodna, 2016-09-26:
Good question; it disappeared when the Docker execution completed. I will re-run and find out for you soon.

mojodna, 2016-09-28:
It's a list of just the JPGs (no `_original`s), as one would expect, so that theory may be bust.

dakotabenjamin, 2016-09-28:
OK. It's probably some command that is ignorantly querying the whole directory. I'll start looking for it.

dakotabenjamin, 2016-09-28:
OK I think I found the problem and/or a potential solution. Lines 82- in scripts/odm_georeferencing:

```
            kwargs = {
                'bin': context.odm_modules_path,
                'bundle': tree.opensfm_bundle,
                'imgs': tree.dataset_raw,
                'imgs_list': tree.opensfm_bundle_list,
                'model': tree.odm_textured_model_obj,
                'pc': tree.pmvs_model,

```

imgs takes the images folder rather than the images_resize folder. I'll create a PR shortly, can you test?

mojodna, 2016-09-28:
Definitely. It'll be tomorrow / overnight.

closed 2016-09-29 by dakotabenjamin

smathermather, 2016-09-30:
We're not loosing any resolution by doing this, are we?

dakotabenjamin, 2016-10-03:
Maybe we should look at #364

dakotabenjamin, 2016-10-05:
I'm considering reverting this change in favor of a more rigorous image verification.

For example, it might be a better idea to ask users to specify their images directory (`-i /path/to/images`) and then copy valid images over to a project directory (which could be the same), rather than force users to adhere to our project structure.

Bavar2142, 2016-10-06:
From an  end users perspective thatd be appreciated.

dakotabenjamin, 2017-06-27:
Changing to resized has repeatedly come to bite us, see #541 #611 I'm looking into a better solution